arkypita/LaserGRBL #262: LaserGRBL custom buttons backup...!

TehseenHasan, 2018-02-01:
Please add option to export and import custom buttons with its icons to a file... It will be very helpful when I install LaserGRBL software to another PC...while now I have to re-write custom buttons again with its custom icons every time I re-install software... Actually I'm using LaserGRBL on many computers... Please add this feature... Thanks..!!!

arkypita, 2018-02-01:
C:\Users\YOUR ACCOUNT\AppData\Roaming\LaserGRBL\CustomButtons.bin contains the data of custom buttons. It is not so easy like an import export but if you copy this file from one pc to another... it work.

closed 2018-02-01 by arkypita

arkypita, 2018-03-20:
[CustomButtons - Copia.bin.gz](https://github.com/arkypita/LaserGRBL/files/1828889/CustomButtons.-.Copia.bin.gz)