arkypita/LaserGRBL #263: Plywood parts cutting

jabbott52, 2018-02-01:
Hi arkypita,
Just wondering how you got such good plywood cuts when making parts for your cnc.  I have a 2.3W laser and can't seem to get all the way through a .25" piece with many passes at full power.  I tried moving Z down a little on each pass for refocusing but still can't seem to get all the way through.  Perhaps the type of wood? Any details on how you made your parts would be appreciated.

Thanks for your fine work!

arkypita, 2018-02-02:
With my 2W laser diode i can cut plywood of 5mm (0,20") in 8/10 passes at a very slow speed (about 200mm/min) with a very sharp result, without moving z (i don't have z). I think the power of the laser matter at least as the **quality of the lenses** and a good focusing.
Also beware of too cheap china laser... declared power doesn't match real power!

jabbott52, 2018-02-05:
Slower works much better.  Thank you.

closed 2018-02-05 by jabbott52