arkypita/LaserGRBL #264: LaserGRBL Configuration wierdness

gerhardgoosen, 2018-02-01:
Hi I have a question? I was using a Chinese nano clone running GRBLv0.9j to control my machine, it was working. I then plugged in a uno (loaded with GRBLv1.1f) with the https://blog.protoneer.co.nz/arduino-cnc-shield into my pc and opened with laserGRBL.

Now when I plug the v.09 board in and try to command it with laserGRBL i get an Error.... the config it shows me in the application is that of the v.1.1 controller but when i type $$ it displays correctly v.0.9 setting values.

i tried un-installing laserGRBL and re-installing it... but still it shows the v1.1 configuration values under menu - > Grbl -> Grbl Configuration...

how do i clear this cached values?

Thanks
I love your APP  =)

arkypita, 2018-02-01:
Yep! you are right.
LaserGRBL shows correct unit and descriptions ONLY for v1.1, with older version name, description and unit are wrong. However numbers and values is correct.

This is a known bug, i will fix it as soon as possible https://github.com/arkypita/LaserGRBL/issues/224

gerhardgoosen, 2018-02-01:
Thank you.. so is there a way i can go back to pre 1.1 config

arkypita, 2018-02-01:
Sorry.. the bug I mention is only with v0.8
To update the grbl config in LaserGRBL when switching between two controllers simply open the grbl conf form and press "refresh" button.

[image]

closed 2018-02-08 by arkypita