arkypita/LaserGRBL #266: some suggestions

timofthedeep, 2018-02-07:
I would like to see a single command button and setting for soft limits  and color change in the graphic where the tool has been in the future?  Thank you.

arkypita, 2018-02-08:
request is not clear, please reword your question

timofthedeep, 2018-02-08:
Soft limits- install a setting that has a maximum distance from home so the application does not try to move farther than the physical limits of the axis without limit switches.  For example If the physical limit of the machine is 150mm, the setting could be input by the user to move 149mm maximum.
Single step button- click button and application sends one line of g-code to machine.
color change- after the application sends a line of code to machine, change color on display graphic to indicate the move has been processed.

arkypita, 2018-02-13:
- Soft limits is already implemented in grbl firmware, just [enable and configure it](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#20---soft-limits-boolean). No needs to implement in LaserGRBL because it is difficult to control each source of possible move and to know the real position/final destination of code.

- Single step button is nice idea, but who and why should use it?

- Color change [already asked](https://github.com/arkypita/LaserGRBL/issues/186) and in roadmap, not so easy to implement.

closed 2018-02-13 by arkypita