arkypita/LaserGRBL #267: Help Wanted

DJCreations, 2018-02-07:
New to This, just purchased New GRBL and Installed according to setup instructions, but system (Laser) activates once turned on,  then nothing ??  is there any type of Remote Assistance provided or available ??

respectfully  Doug
D-n-J-Creations@outlook.com

arkypita, 2018-02-08:
contact who sell your system

DJCreations, 2018-02-10:
I got it from Supporting and Donating for the original Build, and from the POC (Point of Contact)  Paul de Groot <paul.degroot@healthdirect.org.au> but since the delivery, I have sent multiple message with no reply’s, I seen the Gerbil Software Program you have and wanted to incorporated it from you, but that doesn’t seem to work either ??

I found someone through a general GRBL forum who guided me to “LightBurn” which has me at least operating, My other 3 systems will try another Controller and Software so I can at least be provided some type of support, to include the couple of New systems we are getting ready to purchase.

Sorry if I reached the wrong person here as well. Was just following the blogs and reviews, but I will provide respectful comments so no one else follows the wrong path I evidently followed

RESPECTFULLY  and Best of Luck : Lawrence Kelley

arkypita, 2018-02-12:
Where do you purchase your system? Can you provide a link?
Whould you like to upload some pictures of your driver board and laser?

closed 2018-02-27 by arkypita