arkypita/LaserGRBL #268: After the change of laser, the gradation disappeared

foxrus87, 2018-02-07:
Greetings. I changed my 0.5W laser to 2.5 and now I can not adjust it for engraving photos. Can you help me?
On a weak laser, photos were obtained (but that was a long time), therefore I decided to buy a laser more powerfully, but in the end it never happened to draw at least one photo beautifully ...

arkypita, 2018-02-08:
Look like a problem with your laser pwm/ttl modulation. Check your connections and test your driver/laser module.

foxrus87, 2018-02-09:
The laser module is working, the problem is in regulating its power.
but anyway, thanks for the advice.

arkypita, 2018-02-09:
What is the value of $30, $31, $32?

[image]

What value did you use in raster import S-MIN/S-MAX?

[image]

foxrus87, 2018-02-09:
30-1000
31-0
32-1
Smin-0
Smax-1000
So I had with the first laser (0.5W)
I also tried the values 30-255 31-0, but it's still the same.
That is, for example, with a value of C from 0 to 200 glows dimly, and then
immediately and sharply, strongly.
At work, it was suggested to change the circuit between the Arduin board
and the transistor in charge of capacity regulation. place a resistor ...

hk89, 2018-02-09:
That sounds it could be a problem with the driver that came with your new laser. I had a similar problem in the past. Try changing the timer prescaler 1/8 of the TCCRB to get a more slow PWM. Default is at around 7.8 kHz and can go down to 0.98 kHz. Details are on the grbl repository, not here. Google is your friend.

foxrus87, 2018-02-09:
Many thanks for the advice! Good luck to you

closed 2018-02-27 by arkypita