arkypita/LaserGRBL #271: Problem with live visualization of laser pointer running

withoutmessage, 2018-02-08:
A minor issue, somehow annoying: I am having trouble with the live visualization of the laser pointer while it is running. For a while it was ignoring the lower margin, so it would start at the very bottom of the screen. Now it ignores all the left side of the interface as well, starting at the bottom left and drawing upwards appearing and disappearing intermittently from the preview. Let's say, not matching the preview of the image to be cut/engraved.

Thanks!

arkypita, 2018-02-08:
Is this changed with the latest update (from v2.8.35 to v2.8.36) or do you have this issue also in previous version?
I have made a smaller change to draw it bigger to improve cursor visibility, but i don't think it matter

withoutmessage, 2018-02-08:
Hi. I have only tried the last version on Windows 10 Pro.
I think you are right: the visibility of the cross is not the problem for me but the way it appears off the drawing, not where it should. It is a bit hard to explain, I hope this makes sense?

arkypita, 2018-02-09:
Is your grbl configured for report inches?

$13=1 | Report inches, boolean

withoutmessage, 2018-02-09:
Nope. Indeed, when $13=1 the cross dances in the very bottom left corner of the preview. With $13=0 this is what I get: https://streamable.com/bkeat

The cross starts below the design

StuartB4, 2018-02-09:
Are you using LaserGRBL to create the GCode or another app?

withoutmessage, 2018-02-09:
Hi @StuartB4
I am only using LaserGRBL to create the G-Code. And it works fantastic for the most part.

StuartB4, 2018-02-09:
Ok, that should all be ok then. Just a thought.

arkypita, 2018-02-10:
Do this test and post some screenshoot of your result.

- load a picture from your pc
- set line by line as processing
- set the size you want and add a little offset of X10 and Y20
- generate your gcode and see the preview in main window

Now you will see some dotted ruler/lines on your frame that at the border show the position of your image. Assume 10mm in X an 20 in Y as setted.

Do not run the program, insted try positioning the laser head manually sending

G1 X10 Y20 F500

Take a screenshot of the whole window and post here. Where is the cursor? What numbers are reported as position in LaserGRBL window?

arkypita, 2018-02-10:
Try with this image

https://thumbs.dreamstime.com/b/bright-rainbow-colored-rectangle-mosaic-seamless-pattern-abstract-colorful-background-42654694.jpg

withoutmessage, 2018-02-10:
Hi back. Unfortunately, the problem remains: https://pasteboard.co/H70W4CX.jpg

arkypita, 2018-02-10:
You crop the screenshot just to not show me something important that I asked you!

> Take a screenshot of the **whole window** and post here. Where is the cursor? **What numbers are reported as position in LaserGRBL window**?

arkypita, 2018-02-10:
I also suggest to update to latest LaserGRBL version: https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.36

withoutmessage, 2018-02-10:
I was in a rush when I read your message. I am just trying to help, man.
It works now, with the new version:
https://pasteboard.co/H71hdmM.jpg
Thanks

arkypita, 2018-02-10:
Good! Simply older version was not supporting working offset :-)

closed 2018-02-12 by arkypita