arkypita/LaserGRBL #273: minimum laser requirements ?

papa0urs, 2018-02-10:
Hello everyone,
A mega newbie here is starting the laser engraver project... I'm planning to use the laser diode extracted from one of the DVD players. I'm just wondering if this kind of Class 1 laser is enough, or should I get a Class 2 ?

Thank you for your answers !

arkypita, 2018-02-10:
For engraving leather you need at least 250mW of power, 1W for wood. [Here](http://elabz.com/laser-diode-power-output-based-on-dvd-rrw-specs/) you can find a list of speed vs laser power for DVD burner (not player).

Of course you should build a proper current controlled/limited driver, or your diode will burn immediately if you power it at constant voltage. And of course you should provide a proper heat-sinking and lens focusing system. Focusing is very important for engraving.

Much more simple is to buy a ready to use laser engraving module from ebay/aliexpress.

Be care of your eyes. Direct or reflected rays can blind you instantly.

papa0urs, 2018-02-10:
Thank you very much for all these precious infos. As it turned out that my 2nd dvd player has a plastic tray, I might try to find a dvd burner. But the entire engraving module seems a good idea too.

closed 2018-02-12 by arkypita