arkypita/LaserGRBL #274: Laser on/off and PWM should accept M106 M107 G Code commands

mrgithub, 2018-02-10:
I have a laser diode and converted my Wanhao i3 Duplicator Plus (Marlin compatible i think) into a laser engraver and cutter.

These commands control the laser diode when its using my extruder fan PWM.
M106 = off...also M106 S255
M107 = on...also M106 S0

The laser power can be varied with M106 Sxxx.

M106 S254 is very dim and M106 S0 is full power.

Can these be included in next iteration?

arkypita, 2018-02-12:
I see that the user @chantling is working on a fork of LaserGRBL compatible with marlin.
https://github.com/chantling/LaserGRBL

tater1337, 2018-02-18:
I have a laser on my CR-10, but the codes are reversed (M107 off, M106 S255 full power)

I assume there is a 'save Gcode to file' option for those of us using CD cards?

nice job @chantling, go you!

sinfocomp, 2018-02-26:
Hello, why not add a setup option to let the software know wich code tu use to turn on laser, wich code to use to turn off laser and the sxxx max and sxxx min values and whether s is normal or inversed, kind of a post procesor mod for the generated gcode from the software

closed 2018-02-27 by arkypita

sgsalman, 2019-05-24:
@arkypita most important thing m106 m107 codes I think. Please add them :/

arkypita, 2019-05-26:
Contrary to what one might think, this change is not as simple as replacing a string. There are various points in the code that are affected by such a seemingly simple change:

- interpretation of the gcode and preview
- calculation of timing
- state building and recovery operations after abort
- automatic optimization of the generated code (M3 / M5 are modal and can be optimized, other commands are not) and more.

ETE-Design, 2020-05-30:
@arkypita Has the M106 Sxxx + M107 Sxxx been implemented yet? Can see people have asked for it for ages...

DeathDealer33, 2020-12-30:
> Contrary to what one might think, this change is not as simple as replacing a string. There are various points in the code that are affected by such a seemingly simple change:
>
> * interpretation of [...]

Fr i had to write a script to post process gcode from another program just to add this. It is actually simple to do.. Not sure why its not done yet. Ive even seen a version of it with those commands but i cant seem to find it..