arkypita/LaserGRBL #275: Power setting question

edcasati, 2018-02-11:
I'm confused as to the power settings for laser.
Does it go from 0-256, or 0-1024 or what is the range?
What number would be full power?

arkypita, 2018-02-11:
Hi, the arduino pin 11 used for PWM modulation is 8 bit 0/255 different value.
However grbl is able to use a different virtual range that can be [configured with $30](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#30---max-spindle-speed-rpm) and will be mapped to 0-255

edcasati, 2018-02-11:
So the range is set by $30 by the GRBL config.h file, and it seems to be set to 255 by default.
Is there a way to run a startup script in lasergrbl, maybe upon reset, to run some commands, like setting $30, $32 (Laser mode), or other settings that we may want defaulted differently?

hk89, 2018-02-11:
Please read grbl official documentation https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

It says very clearly:
> $x=val - Save Grbl setting

The $x=val command saves or alters a Grbl setting, which can be done manually by sending this command when connected to Grbl through a serial terminal program, but most Grbl GUIs will do this for you as a user-friendly feature.

closed 2018-02-27 by arkypita