arkypita/LaserGRBL #276: Problem with hard limits.

MXguy1000, 2018-02-12:
I know this topic has been beaten into the ground but I have yet to see my issue listed anywhere.. (most are line noise issues. (I wish I could get that far))
So here goes.
Once I enable hard limits with $21 - 1 it trips the alarm right away.. setting $5 to 1 has no effect.
I'm using 2 switches in series (per axis, with Z jumpered because I have no Z axis) in NC configuration.
Even using a jumper to short the X,Y,Z, limit pins straight to ground does nothing.
Is there another setting that I'm missing?
Help..

arkypita, 2018-02-13:
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

arkypita, 2018-02-13:
https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#homing-cycle-isnt-working-right-the-movements-are-all-going-in-wrong-directions-or-its-not-stopping-when-hitting-the-limit-switch

MXguy1000, 2018-02-13:
Guess I just wont use limit switches. read those pages 3 days ago..

arkypita, 2018-02-13:
For NC configuration $5 should be = 1 and a restart (power off / on) af arduino is needed to apply this configuration. Are you sure of what pin you have used, expecially for Z limit?

Between grbl v0.8 and v0.9 they have swapped Z-limit with spindle enable to use hardware PWM for spindle. Some older motor board use the old mapping/serigraphy

https://github.com/gnea/grbl/wiki/Connecting-Grbl

Did you use an RC network to limit noises? It is very important. In my setup I had a lot of problem with limit switches, without RC. My setup is two switch per axes in NO parallel configuration, with pull-up resistor and filtering capacitor.

MXguy1000, 2018-02-13:
I've done the $5 -1 and power cycled.. no effect.. Pin 11 is PWM, 12 is Z and noise isn't the problem, even with everything but the arduino off it still triggers the alarm instantly.. Once it arrives and the limits are working I will be using an opto coupler to stop noise.. I would rather have NC because then if there one of the wires breaks or a switch goes bad it will trip the alarm, and I won't be running a machine thinking the limits are working only to find out the hard way that they are not..

MXguy1000, 2018-02-13:
I'm going to try no switches with $5 - 0 and hard limits on.. that should at least not trip the alarm.. no switches = NO..

arkypita, 2018-02-13:
Your setup sounds to be correct. The approach on NC for safety is a good approach, also NC to ground should be more noise resistant.

I cannot guess what could be your problem. What happen if you send $X?
Did you have homing enabled? What happen if you send $H?

With my first try with limits switches i had problems too, but i am not so experienced to know all the possible problems. My problem go away with pull up resistor and filtering capacitor.

Good Idea to try without switched and NO, just to see if it works.

Maybe you could have a damaged input pin on your arduino board?

arkypita, 2018-02-13:
However please note that the correct github for this kind of problems is the grbl one, where you can find grbl developers and experienced users, and a lot of already asked questions about homing and limit switches. LaserGRBL is only one of the frontend for grbl.

https://github.com/gnea/grbl/issues?utf8=✓&q=is%3Aissue+limit (v1.1)
https://github.com/grbl/grbl/issues?utf8=✓&q=is%3Aissue+limit (v0.9)

MXguy1000, 2018-02-13:
It is starting to behave better.. It might have been that clearing the alarm is harder that it should be.. Needing to unlock and reset GRBL.. I'm guessing that I won't be able to use the homing feature because I don't have a Z axis (and that is the first axis it tries to home)? And thanks for the links to the proper rant rooms..
Thanks again for your time..

arkypita, 2018-02-13:
You can do a small change in grbl source code to disable Z axis homing.
It's so simple as comment one line and deccoment another one.

Look for file [config.h](https://github.com/gnea/grbl/blob/master/grbl/config.h#L90) in your arduino library folder, and edit it. Comment line 105, 106 and decomment 110. Recompile grbl and upload to arduino.

MXguy1000, 2018-02-13:
Thanks.. I think however.. I will just use Z limit as a E-Stop for now, I do plan to add a Z axis so I can focus the laser at different heights. I have a 50mm Z axis on the way now.. That and I didn't (so far) get the source code for GRBL I just downloaded the .hex file..

MXguy1000, 2018-02-13:
So.. to sum up what my problem was.. (hanging head in shame)...
In order to clear an alarm you must first reset, then unlock, then back off the offending switch..
And you WILL need to repeat till the switch is no longer tripped..
Thanks again..

closed 2018-02-13 by MXguy1000