arkypita/LaserGRBL #277: Won't connect to Arduino Clone

powerhi, 2018-02-12:
Trying to connect LaserGRBL to Arduino Clone (DCCduino). Got the requisite drivers(from [here](http://www.kupply.com/dccele-dccduino-arduino-uno-clone/)) for the clone and I'm able to upload to it from the Arduino IDE. However when trying to connect the microcontroller to the application, it says connecting for a few seconds before eventually stating disconnected. The log file for the operation is as follows 

1:34:05 PM	SetStatus   	Machine status [Connecting]
1:34:05 PM	OpenCom     	Open COM3 @ 115200 baud
1:34:05 PM	SendImmediate	Send Immediate Command [0x18]
1:34:15 PM	OpenCom     	Connection timeout!
1:34:15 PM	CloseCom    	Close COM3 [CORE]
 1:34:15 PM	SetStatus   	Machine status [Disconnected]

I will say that I've tried connecting before and after adding a picture and it still doesn't work. What piece here am I missing? What am I doing wrong?

arkypita, 2018-02-12:
Did you have flashed grbl firmware to your arduino?
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino

powerhi, 2018-02-12:
What hexfile do I supply XLoader with?

arkypita, 2018-02-12:
https://github.com/gnea/grbl/releases

arkypita, 2018-02-12:
You are welcome

closed 2018-02-12 by arkypita