arkypita/LaserGRBL #283: Scan-Dot mode

npelov, 2018-02-19:
The best image quality I've produced until now is with EleksCam's Scan/Point mode. It's like your line to line tracing, but uses series of dots. In line to line mode the start and stop of the line is usually burned more than the rest of it. Dot mode is really slow, but it produces nice even burning. 
In dot mode the most important parameters are laser power and time to leave the laser on in ms. And of course dots per mm (it's one value for both horizontal and vertical).

Their G-code looks like this:
G0 X1.234 Y1.234 - position
M3 - turn the laser on (although they use S code instead)
G4 P0.001 - wait 1ms
M5 - turn the laser off

The G4 Pn.nnn command's resolution is 1 ms (P0.001, although I don't think my hardware can achieve it.)

arkypita, 2018-02-23:
Did you try LaserGRBL with dithering?

https://www.youtube.com/watch?v=7i9V2N755d4
https://www.youtube.com/watch?v=qU4-vq3Uy4Q
https://www.youtube.com/watch?v=HZ7_dFz6IrI

npelov, 2018-02-23:
Yes, it's a great feature. It works fine for bigger photos with gray levels. But it doesn't work that well with small solid black and white graphics. It still draws lines, not dots which makes them faster, but you loose detail for very small images - like engraving pens, key chains, etc. If you set higher lines per mm the burning in the solid area gets a lot deeper than the thinner areas. If you set lower, the filling doesn't look good. Dot mode works very good for text.

I would use EleksCam software for small items, but it is unusable if you want to burn more than one image on the same media. It crops white space, so even if the images have the same sizes they do not start from the same place. For example burn text using dot mode and then outline it using a solid object and outline mode.

arkypita, 2018-03-10:
Nice hint, i will try to implement this mode in future.

closed 2018-03-10 by arkypita

MartinW2, 2022-03-04:
Strongly supporting this request! 👍 It seems to be only few steps away in priority list :-)