arkypita/LaserGRBL #284: LasrGRBL non mi funziona più sulla Eleks Maker A3 Pro

brillo9772, 2018-02-20:
Salve, come avevo scritto nel titolo, da qualche giorno, dopo aver fatto l'aggiornamento alla V. 2.8.36 ha iniziato a perdere la connessione tra il software e l'hardware, provo spiegarmi meglio come mi è successo. Inizialmente mi è capitato alcune volte che durante l'incisione si fermi e non si riesca più a dare comandi da software, come se mi avesse crasciato il programma. Rispegnendo e ricollegando il tutto, varie volte, si riesce a ripartire per un po', ma con il medesimo epilogo, di bloccarsi tutto, questo sempre più frequentemente, fino ad arrivare al punto che quando vado a collegare l'incisore al software, tramite pulsante dedicato, mi dice che "Un dispositivo collegato al sistema non è funzionante".
Visto che la scheda dell'Arduino nano e una compatibile, ho provato da Arduino IDE a ricaricare gli sketch ma mi da errorore, è possibile che si sia danneggiata la scheda? Grazie per l'attenzione.

arkypita, 2018-02-20:
Ciao, tra la versione 2.8.36 e la precedente 2.8.35 non ci sono modifiche che riguardano la comunicazione. Ad ogni modo carica il file di log (menu ? - open session log) così ci do un occhiata.

Il fatto che non riesci a caricare gli sketch potrebbe essere dovuto al fatto che le EleksMaker non hanno il bootoader? Hai mai caricato gli sketch su quel nano oppure no?

Dai sintomi che descrivi sembrerebbe che hai problemi seri di comunicazione, cosa che capita di frequente con i cloni di arduino che usano il chip CH340. leggi le [faq](http://lasergrbl.com/faq/) e prova le soluzioni consigliate.

arkypita, 2018-02-20:
ps. la prossima volta meglio se scrivi inglese, le tue domande e le mie risposte possono essere utili anche per altre persone.

brillo9772, 2018-02-20:
Grazie per la rapida risposta, per quanto riguarda l'inglese, mi dovrò appoggiare a un traduttore..
Thanks for the quick reply, as regards English, I will have to support a translator ..

arkypita, 2018-03-10:
hai risolto?

closed 2018-03-20 by arkypita