arkypita/LaserGRBL #285: Laser Not enough power

THEDOOMAN96, 2018-02-22:
Hey so I am new to this program and I would like to start using it because the software my engraver came with was a bunch of junk. I bought a bigger laser that is 2500mw and this is the engraver I own: https://www.amazon.com/gp/product/B01M0QHAF7/ref=oh_aui_detailpage_o09_s00?ie=UTF8&psc=1

I get LaserGRBL to print, however the laser is not leaving anything on the wood. The laser is not near as bright as it is on the other software so I know its not the laser. Is there a setting that I am missing that will give the laser more power to start actually leaving marks on the wood?

arkypita, 2018-02-22:
Hi
Laser power is PWM modulate by arduino pin 11. Native modulation range is 0-255 as arduino has 8bit PWM modulation.

Grbl add a virtualization on 0-255 to allow you to use a different range (i.e 0-1000) and scale it to 0-255. This virtual range is [configured via $30](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#30---max-spindle-speed-rpm)

LaserGRBL control laser power sending S+VALUE command (like S123, S255, S0) scaling the grayscale tones of the image, using min/max settled in raster import.

So, if your $30 is 1000, you should use Smax of 1000 in raster import to use the max power.

THEDOOMAN96, 2018-02-22:
okay that makes more sence. I will give that a shot when I get home from work and i'll let you know.
Thank You

closed 2018-03-10 by arkypita