arkypita/LaserGRBL #286: GCODE with M106 and M107

Fidosup, 2018-02-22:
Hi dear. First of all my congrats for your magnificent work.
I think to user like me, with a 3D printer, it might be very useful if the software let us change the gcode or instruction to manage the laser pointing to the fan port ( M106 and M107 ) as Inkmaster does.

Do you think it is too much work or it is possible ?.

Cheers and thank you.

Mauricio Torres.

sinfocomp, 2018-02-22:
I think you could do this in your post processor

arkypita, 2018-02-26:
https://github.com/arkypita/LaserGRBL/issues/274

closed 2018-02-27 by arkypita

7oRk3, 2022-07-13:
hola amigo soy nuevo. y compre un modulo laser para mi creality ender 3 que funciona con m106 para prender y m107 para apagar el laser. veo que hay varias entradas que lo solicitan y ademas que lo marcas como solucionado pero de verdad no e podido ubicar la version que me funcione o que al momento de selecionar donde aparece m03 y m04, que aparezca tambien m106 y m107. puede ser que mi solicitud sea tonta me podrias dar el link de la version que posea asi y no se si tengas una que exporte el codigo a .gcode para colocar directo en la impresora 3d crality ender 3 mil gracias de antemano