arkypita/LaserGRBL #291: A lot of disconnections

powercreations, 2018-03-05:
Hello everybody, i try to use laserGRBL in my clone K40 laser with smoothieboard  and grbl LPC firmware as i really like it but i have many disconnections.

I change cable (short one), i try usb3/usb2, i test with speed slow/fast, i try with pixel per mm (10-6) but nothing fix it. 

I use other program and works ok.

What you think i do wrong? Thanks in advance

arkypita, 2018-03-05:
can you provide the program log?
menu ? - open session log

powercreations, 2018-03-06:
I find session log, where is the rest? :) 

I make a restart of pc and look like that was working ok but i will make some more tests today to see.

[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/1784103/sessionlog.txt)

arkypita, 2018-03-06:
Session log is automatically trimmed to 1000 lines
The exception listed in log show that the COM port disappear/disconnect in windows, I cannot guess what could be the cause, maybe conflicting with other software/driver or cable problem.

You can see the com port if exist/disappear looking into device managment
https://support.microsoft.com/en-us/help/4026149/windows-open-device-manager

powercreations, 2018-03-06:
Thanks for your help, i would check how is going in a while,

Is any way to connect it with smoothie ethernet?

arkypita, 2018-03-06:
> Is any way to connect it with smoothie ethernet?

I have developed a telent connection that allow to send/receive gcode using simple TCP/IP telnet protocol but i don't know if it work with smoothie. Try yourself

[image]

[image]

powercreations, 2018-03-06:
It looks that it work ok, after laserGRBL setup i had a lot of disconnections but after of pc restart works great.

I will try and inform you if i manage to make it with ethernet, thanks again.

closed 2018-03-06 by powercreations