arkypita/LaserGRBL #293: теряет координаты

Vytelis, 2018-03-09:
почему при гравировки фотографии,теряет координаты и курсор уходит за пределы картинки?
[image]

arkypita, 2018-03-10:
The usb-serial communication between PC and Arduino does not use any sort of error-correction or error-checking algorithm, so it is possible that command sent from LaserGRBL/PC could be received slightly different from grbl/arduino. This is intrinsic in this communication type and cannot be fixed by software. https://github.com/grbl/grbl/wiki/Known-Bugs#usb-to-serial-transmission-errors

LaserGRBL contains some communication slow-down that help to minimize this problem, but the best correction is to have good quality USB cable, shielded and far away from motor noise, grounded circuit, and powerline decoupling capacitor.

http://lasergrbl.com/faq

vovpff, 2018-03-14:
@Vytelis попробуй так же уменьшить ускорение и/или фидрейт.

closed 2018-03-20 by arkypita

Vytelis, 2018-03-22:
Уменьшил ускорение и фидрейт,подключил через качественый кабель,то же самое.Сейчас спомощю Lazer GRBL генерирую G-код, его вставляю в GRBL Controller,а с его помощю работает отлично.

vovpff, 2018-03-22:
@Vytelis версия grbl какая? лазер в режиме шим работает?
В lasergrbl открывай менбю grbl -> настройки. Пробуй разные варианты Streaming mode (хз как по русски)

Vytelis, 2018-04-05:
С новой версией laser grbl v2.8.40 работает нормально.