arkypita/LaserGRBL #294: Path following issue

AntoineKia, 2018-03-09:
Hi, :) 

I have a weird problem since 2 days with the software.

https://youtu.be/dm44c5HOyN4

GRBL configuration : https://imgur.com/a/bkkmH

As you can see in this video : the shaps of "closed path"/circles are correcly followed by the pointer/laser but the relative positions of these "closed path" arn't at the good location and the final result is a huge mess.

Because the blue pointer is also affected without asservissement, I think this is a software problem.

Please help :(

Thanks,
Antoine

AntoineKia, 2018-03-09:
Changing threading mode or transmission mode doesn't fix the problem :/

arkypita, 2018-03-09:
could you share your gcode file?

arkypita, 2018-03-09:
I have configured my grbl same as your, and test with a similar file type/size
This is my result, in my opinion there is not a problem in LaserGRBL: https://youtu.be/qG0Ppqip6j4

Imho you have noise on USB cable/line and sent command are received slightly different from sent.

AntoineKia, 2018-03-09:
I'll try to buy an insulated cable. It worked with a shorter one before. 
If it doesn't work, I'll share the gcode.
Thanks for that fast answer :)

arkypita, 2018-03-10:
Other hint:

If you use a laptop plug it to the power line and do not run with battery. Doing this you have grounding through main power line.

Place your USB cable far from motors and DC line, to minimize induced noise.

Add a pair of capacitor parallel to DC line, one elettrolitic of 47uF 50v and one ceramic of 0.1uF. this is called “decoupling” and could help in case of electrical noise.

AntoineKia, 2018-03-14:
It worked with a shorter cable (but not with the 1.8m insulated one) :) Thanks for your support !

closed 2018-03-14 by AntoineKia