arkypita/LaserGRBL #295: Unidirectional engraving produce wrong result

arkypita, 2018-03-10:
I found an issue with vectorize+filling when used with the flag of unidirectional engraving:

[image]

The same image with the same parameters, created without unidirectional flag produce correct output:

[image]

Citizen63, 2018-03-17:
Firstly, congratulations and thank you for your fine work. It has provided me with a great deal of pleasure and satisfaction over the past while, during which time I have built a diy laser rig. Your software is by far the best I have found. Cheers.

However.......

I agree with your findings that there seems to be a bug in the rendering side of things. I have thought this for a few months now. With the programme configured for unidirectional engraving, the error
 you describe occurs, Not only filled vectors, Line to Line Tracing is also affected. This issue affects all releases since the unidirectional option was introduced, as far as I can ascertain. It's a real pity as I rely heavily on the unidirectional functionality, as the backlash on my DIY rig is unbearable! (Selfish, I know).

Bidirectional:
[image]

Unidirectional:
[image]

[image]

[image]

Here's hoping you manage to find time to give this issue some investigation.

Thanks again.

arkypita, 2018-03-17:
I have already found the cause, but the solution is not so easy to implement.

In a first pass i look for black and with pixel sequences, generating "segments" that are lines of consecutive pixels of the same grey tone with their length (integer, number of consecutive pixels).

In a second pass i transform all that segments in gcode scaling number of pixels to real mm, producing a double number with a fixed precision (3 decimals) and this necessarily introduces a rounding error.

When in bidirectional engraving segments are produced in both direction, and this balance rounding errors. When unidirectional I produce segments only in engraving direction, than I put only one big segment to "go back". The big segment introduce only a small rounding error, while all the small segments introduce a cumulative rounding error.

If I use two decimals the error is huge, with 3 decimals (actual setup) is visible, with 5 decimals it disappear.

Two decimals
[image]

Five decimals
[image]

Citizen63, 2018-03-18:
Thanks for your prompt reply. I found it informative but it makes me think of a few questions.

> I have already found the cause, but the solution is not so easy to implement.

You seem to have implemented the solution for the example you showed above.

1: Is this something I too could do, if so, how?  Please refer me to any relevant tutorials.  I have basic programming skills, just enough to get me this far, but eager to learn.

2: Is there a reason that calculating to five decimals is not the norm? Does doing so adversely affect other aspects of the programme?

3:  Would you consider a  future release that uses the five decimal place method.

Thanks for your time,
Cheers

arkypita, 2018-03-19:
> You seem to have implemented the solution for the example you showed above.

The solution I implemented is only one line of code that change the gcode output from #.### to #.###
in [GrblFile.cs line 96](https://github.com/arkypita/LaserGRBL/blob/1501dbaa4bde66893144a041ce52fa5d5ddddda3/LaserGRBL/GrblFile.cs#L96)

However, this so simple solution it is not the right one: rising up the number of decimals will increase the number of character to be sent to grbl, with the result of slow down the streaming speed and limit the maximum engraving speed. For someone this could not be a problem, but for others could be a limit.

The best solution is to cumulate the rounding error to a temporary variable and keep it counted for each next step of calculation. I am trying to implement this way.

arkypita, 2018-03-19:
Something like this:

`			static double errX;
			public string formatnumberX(double number)
			{
				//return number.ToString("#.###", System.Globalization.CultureInfo.InvariantCulture);

				double orig = number + errX;
				double send = Math.Round(orig, 3);
				errX = send - orig;
				return send.ToString(System.Globalization.CultureInfo.InvariantCulture);
			}

			static double errY;
			public string formatnumberY(double number)
			{
				//return number.ToString("#.###", System.Globalization.CultureInfo.InvariantCulture);

				double orig = number + errY;
				double send = Math.Round(orig, 3);
				errY = send - orig;
				return send.ToString(System.Globalization.CultureInfo.InvariantCulture);
			}`

arkypita, 2018-03-19:
I have released a version that fix this issue: https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.37
@Citizen63 @Dpsspower and any other user that is using unidirectional can test? Also test bidirectional please :-)

Citizen63, 2018-03-19:
Good morning, thanks again for your time. I will give your latest release good workout tonight when I get home from work.

Dpsspower, 2018-03-19:
Hi,
To be honest, I never saw this issue because I never engraved something using the outline vectorisation feature. I do all vectorisation using separate picture processing software.
In the line by line mode, this error never occured to me. I used an older version from end of last year until now. Never change a running system. I never had such distortions in line by line mode.
I always use the unidirectional mode. And it does its job great, increasing sharpness of results dramatically! :-)
I checked it using outline vectorisation + fill and yes - the error was there. But never in line by line only mode.
So this seems to have something to do only with the combination of vectored outline + fill, not with simple line by line engraving.

Now I installed the new release and the error seems to be gone on the screen.
I agree that calculating the error "on the fly" is the better way than increasing output resolution of the output code.

The only thing I cannot understand is, why the problem just happened when adding both outline and line by line fill, and not when using line by line only only.

I will do some more tests.
Thank you very much.

arkypita, 2018-03-19:
The issue is in line2line algorithm I use both for lin2line and vectorized filling, but when you use vectorization+filling it's more obvious because you have a visual reference.

[image]

It is also more visible with certain resolution settings, I.e. with 8 line/mm seem the problem disappear, but with 7 line/mm is very strong. 

[image]

[image]

arkypita, 2018-03-19:
With the last 2.8.37 the problem seem to be minimized, but not really solved:

[image]

[image]

arkypita, 2018-03-19:
Please test with v2.8.38, should be the final one!
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.38

Citizen63, 2018-03-19:
Hi

Awesome.
I didn't get a chance to try 2_8_37 but I'm putting 2_8_38 through a couple of tests.

(20 minutes later).....

I can't break it! 
Pin sharp imaging now. I'm going to get my goggles and fire up the laser. 

I'll get back to you later after I check some results.
Congratulations and thanks.

closed 2018-03-20 by arkypita

Citizen63, 2018-03-20:
Roger
Over and out.

Jelmandesign, 2024-03-16:
Is it possible in the latest version of Laser GRBL (v6.2.1) to enable and disable bidirectionality?

arkypita, 2024-03-16:
@Jelmandesign nothing has changed, did you have any evidence that the future does not work?

Jelmandesign, 2024-03-16:
I have foundation it in 2e window. Thanks for your reaction. I like your
program!

Kind regards,
Jan Laarman

Jelmandesign, 2024-04-04:
I have another question.

For Laser cutting I use my CNC 3018 Pro with laser unit.
The nice thing about LaserGRBL is the possibility to make extra buttons
your self with the functions you want added. to them
Now I have created a button with the command Go to Home position
(right/Top) and then to Left/Bottom. In my case it looks like this:
$H
$j=G91 X-274 Y-162 Z-24 F6800
$x
$x

This puts the machine (for me) in the starting position.
Before I can lasering an opject is ofcours needed.
After this has been imported it is necessary to comfirm the 0-position
with: G90 X0 Y0 Z0 (Third icon from left (with globe symbol)).
What I would like is to add this function to my "ZERO-Button".
So my question is what is needed to add to my program, so that the
0-position-command waits until the move to Left/Bottom is abyssed. So that
at one push of my button the machine executes the following command:
1. Home
2. Left/Bottom (+Z position)
3. ??? wait for confirmation ???
4. G92 X0 Y0 Z0

I'm Sorry for my bad English.

Jan Laarman
The Netherlands