arkypita/LaserGRBL #296: Deceleration makes sides of print dark

David-Woodthorpe, 2018-03-10:
Hello - would it be possible to add an option to enable full scan passes so that the laser doesn't dwell at the sides of the print area causing darker areas of engraving - perhaps also adding a few mm of white space so edge to edge laser engraving doesn't get this as well?

hk89, 2018-03-10:
Using the M04 instead of the M03 should give you better results. M04 command sets the laser modulation according to the speed the laser is traveling at, including the areas of accelerations and decelerations.

arkypita, 2018-03-12:
good hint from hk89
btw the request is already in development roadmap

closed 2018-03-12 by arkypita

z3nunes, 2019-08-22:
I have also been following your work since the beginning direct use of your program and updates, but always noticed that in the recording mode there are problems with acceleration and deceleration, either with M3 or M4, the ideal would be to add borders on the design for acceleration or deceleration to occur. without recording anything right now..for recording to be cleared. Thanks for the work.