arkypita/LaserGRBL #300: Keep filename when saving a gcode file

therealjedidad, 2018-03-13:
When saving an open file, the existing filename is not persisted to the save dialog. I have a changeset ready for review with this change if I can be granted access for a pull request.

arkypita, 2018-03-13:
feel free to submit pull request, the access is available

therealjedidad, 2018-03-15:
Still new to using git, I've always used tfs/vsts instead. When I try to push my changes I get an error "remote: Permission to arkypita/LaserGRBL.git denied to therealjedidad."

arkypita, 2018-03-19:
I am not an expert of github but in the past some user submit pull request, so it should be possible:
https://github.com/arkypita/LaserGRBL/pulls?q=is%3Apr+is%3Aclosed

ask @Ranthalion or @Effer how do they get it

Effer, 2018-03-19:
@therealjedidad you should push to your repository and then open a Pull Request to @arkypita repo. You can make a pull request directly from github pages. That's the way I did.

closed 2018-04-18 by arkypita