arkypita/LaserGRBL #303: error with big file

sinfocomp, 2018-03-15:
[image]
i get this error with the attached file

StuartB4, 2018-03-16:
No file attached.

arkypita, 2018-03-24:
Hi sinfocomp
the red cross on gcode panel was already reported by another user, but no longer inspected due lack of information (how to reproduce, when it happens ..). 

If you would like to provide more info...

sinfocomp, 2018-03-25:
[image]

[fajo aros caballos 6 ojillos T42P21V8.txt](https://github.com/arkypita/LaserGRBL/files/1845710/fajo.aros.caballos.6.ojillos.T42P21V8.txt)
attached is the file it happened 20 minutes after starting job, no other interaction with software done,

arkypita, 2018-03-26:
I have tested with your file but the job was completed without having the issue

arkypita, 2018-03-26:
If the problem is repeatable by your side i can build a diagnostic release to acquire some info at the moment of error.

When you experience the issue, did you have an error window like this? 
If yes, can you copy-paste a screen and the "detail" text?

[image]

StuartB4, 2018-03-26:
I also tried the file, just changed the .txt extension to .nc
Job completed as it should.
[image]

sinfocomp, 2018-05-01:
[image]

sinfocomp, 2018-05-01:
This is the content of the error msg details:

ee the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at LaserGRBL.UserControls.CommandLog.OnPaint(PaintEventArgs e)
   at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer, Boolean disposeEventArgs)
   at System.Windows.Forms.Control.WmPaint(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ContainerControl.WndProc(Message& m)
   at System.Windows.Forms.UserControl.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5485 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v2.0.50727/mscorlib.dll
----------------------------------------
LaserGRBL
    Assembly Version: 2.8.45.0
    Win32 Version: 2.8.45.0
    CodeBase: file:///C:/Program%20Files/LaserGRBL/LaserGRBL.exe
----------------------------------------
System.Drawing
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5495 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Drawing/2.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8686 (QFE.050727-8600)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System/2.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5491 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Windows.Forms/2.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Core
    Assembly Version: 3.5.0.0
    Win32 Version: 3.5.30729.5420 built by: Win7SP1
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Core/3.5.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5494 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Xml/2.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5483 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Configuration/2.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
mscorlib.resources
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5485 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v2.0.50727/mscorlib.dll
----------------------------------------
System.resources
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5420 (Win7SP1.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.resources/2.0.0.0_es_b77a5c561934e089/System.resources.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

arkypita, 2018-07-05:
Are you continuing to have this problem?

arkypita, 2018-07-05:
Closing this issue, please continue on #354 (same issue)

commit a80a80ae (2018-07-05): more robust code to fix #303
closed 2018-07-05 by arkypita