arkypita/LaserGRBL #304: Saving Job Profile

foxrus87, 2018-03-17:
Good afternoon.
When working with different materials, each time you enter the values of speed, power and size.
Is it possible to create profiles in some way so that when selecting any of them all these values were entered automatically, so that they are not always remembered in your head.
Example:

file profile.ini

nameprofile="photo on the board"
s_min=20
s_max=200
speed=2000
h_size=150
w_size=100

then you can have several different profiles for different tasks

arkypita, 2018-03-19:
I am already doing this in a development branch, but i suspend development cause i have no spare time.
However I agree with your request

foxrus87, 2018-03-19:
I'm sorry, I hope that soon you will find free time, because your program is very good!

closed 2018-03-20 by arkypita