arkypita/LaserGRBL #306: Share your custom buttons Here!

arkypita, 2018-03-20:
The last [v2.8.39](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.39) allow to export custom buttons. If you want to share yours with LaserGRBL community feel free to reply to this post.

Explain your work and don't forget to attach your CustomButtons.gz file!

Arhimed-Ru, 2018-04-12:
[Arhimed_CustomButtons.gz](https://github.com/arkypita/LaserGRBL/files/1904854/Arhimed_CustomButtons.gz)
[image]

arkypita, 2018-04-13:
thanks for your share!

Arhimed-Ru, 2018-04-13:
another button option
bypass (diameter = diagonal) around the picture

[image]
.....
G0 X[left] Y[bottom] F1000
M3 S1  F100
G2 X[left] Y[bottom-0.002] I[right/2-left/2] J[top/2-bottom/2]
M5
G0 X0 Y0 F1000
.....

I hope this works :)

Arhimed-Ru, 2018-04-13:
[image]
[image]

Arhimed-Ru, 2018-04-13:
how to change the scale of the preview??...  move the preview??...

Xillius178, 2018-04-14:
FAN O/OFF Buttons:
Code for FAN ON
M08

Code for FAN OFF
M09

[image]
[image]
[image]
[image]

Xillius178, 2018-04-14:
@Arhimed-Ru

Can you translate the description of the buttons into English?

Thank you

Venkes, 2018-04-15:
[image]

This button draws a rectangular shape around the object. This allows you to put the object on the edge of the burning area.

----------------------------------------------------
G0 X[left] Y[bottom] F3000
F2000
M4 S100
G1 X[left] Y[top] 
G1 X[right] Y[top] 
G1 X[right] Y[bottom]  
G1 X[left] y[bottom] 
M5
G0 X0 Y0 F3000
----------------------------------------------------

bearded1der, 2018-04-17:
Thanks for sharing your config @Arhimed-Ru! It's basically everything I was using, plus some!

@arkypita Maybe we could see about adding a method for changing custom button order in a future release to make things more easily customizable? Drag and drop maybe?

Koctetcav, 2018-04-23:
Прекрасно.  Спасибо большое. Вот только лазер не квключается отдельно кнопкой.

Venkes, 2018-04-23:
С меня включается лазер. Может быть, M3 вместо M4 с вами?

Koctetcav, 2018-04-23:
Лазер включается только при перемещении.  А мне нужно чтоб при простои моторов включать лазер!?

Koctetcav, 2018-04-23:
Я и M3 и M4 пробовал.

Venkes, 2018-04-24:
лазер переходит в нижний левый угол области гравировки, затем лазер создает прямоугольник вокруг области, подлежащей гравировке, а затем выключает лазер и затем переходит в исходное положение. В моем случае лазер включается с M4 (третья строка) и выключается M5 (восьмая линия).
Ps. Надеюсь, Google переведет все хорошо переведенное!

Venkes, 2018-04-29:
@arkypita yes you are right! I'm sorry. I've editted my comments.

marceneirojovem, 2018-05-16:
the cheap laser module from China, can operate only half hour, after what its necessary a time for cold. Is it possible a custom button to make a pause after this operation time ? Anybody can help ?
Thank you

StuartB4, 2018-05-17:
There already is a pause button.
Press the hand to pause and the man to continue.
[image]

marceneirojovem, 2018-05-17:
I think I did the wrong question. 
I want a automatic pause after 45 minutes and a automatic return after a 30 minutes of cooling time.
 Something like a macro in Gcode to automates the process. Excuse me for the bad english
Thank You

marceneirojovem, 2018-05-17:
Thank you Bearded1der . 
I think it would be useful for everyone who has a cheap chinese laser module.

arkypita, 2018-06-14:
For who like custom buttons try the [latest version](https://github.com/arkypita/LaserGRBL/releases/latest)
Hope you love it!

1estrider, 2018-07-31:
For Z-axis on my laser i had to make a code for up:

G91
g0Z[jogstep]F[jogspeed]
G90

and down:

G91 
G0Z[0-jogstep]F[jogspeed]
G90

1estrider, 2018-08-27:
How do you apply .gz file?

arkypita, 2018-08-27:
right click on custom button area and select "import" from menu

https://user-images.githubusercontent.com/8782035/37682555-381d3a84-2c8a-11e8-8231-beebb8241cea.png

fantilator, 2018-09-09:
$X (Reset Machine )
$H (Unlock )
M5 ( M05:Spindle Stop )
G91 ( G91:Incremental programming )
G0 X-510 y-460 Z-53.6 ( go to center of table )
G92 X0 Y0 Z0 ( Make this my work zero point
G28.1 ( to ensure the machine goes back here)
M3 S200 ( M03:Spindle On at 200mW for zeoing my workpiece )

StuartB4, 2018-09-09:
@fantilator.
Interesting, do you have your Home set to the back right hand corner?
Just wondered because these are negative numbers and would move the laser to the left and down the screen away from the usual Home position.

> G0 X-510 y-460 Z-53.6 ( go to center of table )

fantilator, 2018-09-09:
@StuartB4: indeed my homing position is back right corner Z in upper position.
after the homing the laser indeed goes to left and backwards and downwards.

My max laser power is 8000mW, So i have setup in my GRBL settings (arduino code) that S8000 is my maximum. therefor S200 equals 200mW. all programs support this, otherwise I change the G-code itself.

StuartB4, 2018-09-09:
Ok fanilator. Thanks very much for the explaination, now I understand.

Focus3, 2018-10-31:
Good day,

just wanted to share a custom button for homing. What it does is set the idle delay to 1, so it can do the homing. Then when it homes the CNC it sets it to "Zero point" and after that the idle delay is set to 255. The reason why idle delay is usefull is that it can help prevent lost steps. It holds the stepper motors in place until the next step, thus elimination the efect of vibration.

$1 = 1
$H
G92 X0 Y0
$1 = 255

Hope it helps.

Focus3.

dstronginaz, 2018-11-19:
I am new to using an engraver and love what all have posted here. I would like to make a custom button to start the project. The small one up top, is just too small for a touch screen and I end up changing the number of passes.

GSNeko, 2019-02-04:
Thank you all for sharing your custom buttons, they have been very helpful. 
I'm new at this but I've made this button that shows the center of the image for 3 seconds and returns to 0. 
I use it when I need a perfectly centered image on a circular piece. 

G0 X[left+width/2] Y[bottom+height/2]
M4 S1
G4 P3
M5
G0 X0 Y0 F3000

I'm using gcode for the first time, so let me know if there is something wrong.

StuartB4, 2019-02-04:
Hi GSNeko.
Laser does not come on.

Try this.

$32=0
G0 X[left+width/2] Y[bottom+height/2]
M4 S1
G4 P3
M5
G0 X0 Y0 F3000
$32=1

GSNeko, 2019-02-04:
Thanks Stuart, It was working for me until I restarted LaserGRBL. I've solved it as you said. 
I'm editing mine so there are no mistakes.

StuartB4, 2019-02-05:
Ok GSNeko. Thats good, glad it worked.

adnanmitf09, 2019-03-20:
I have translated the tooltip of@Arhimed-Ru's custom buttons to English. Although icons are self explanatory, its a good idea to have tooltips in English if you don't understand Russian. PFA. [https://drive.google.com/open?id=1O1Un14hJvrpjwJR74DQCwqjF1pKyUgj6](LaserGrbl-CustomButtons-Eng-Adnan)

StuartB4, 2019-03-21:
@ adnanmitf09 
There does'nt seem to be anything in the archive you linked to.
It won't extract.

Stankovo, 2019-04-06:
[image]
Я немного переделал внешний вид кнопок, ближе к единому стилю. Могу поделится.

Stankovo, 2019-04-07:
[CustomButtons555.gz](https://github.com/arkypita/LaserGRBL/files/3051837/CustomButtons555.gz)
Мой набор кнопок. Не забудьте настроить под себя лазер.

Bazza1959, 2019-04-14:
can anyone give me an idea how to open these files. 7zip should but won't nor will zipware.  I only want a button to turn on laser at minimal power to see where to align the work pieces and then turn off

Stankovo, 2019-04-15:
Эти файлы открывать или распаковывать не нужно. Просто на панели, где расположены кнопки, на свободном месте, нажмите правую кнопку мыши и в контекстном меню выберите "Import". Выберите файл с расширением *.gz,а затем "OK". Начнут появляться запросы на размещение кнопок, вам нужно просто соглашаться. Когда последняя импортируется все кнопки появятся в панели. 
Данный архив сам распаковывается программой LASEgrbl.

Bazza1959, 2019-04-15:
thank you very much

alexdam92, 2019-05-15:
Hi, it's possible to change positions of every custom button? i Imported the wonderful Arhimed-Ru's custom buttons but i can't cange the positions... 
And for what i can use the zero position? Is not the same as world with hotspot button? (the third default button)
Thanks Alex

Stankovo, 2019-05-15:
Здравствуйте. Кнопки невозможно перемещать. Можно удалить лишние кнопки.

Stankovo, 2019-05-16:
В команде x=0, y=0 нет команды z=0. Позиция Z не обнуляется.

alexdam92, 2019-05-17:
thanks so much Stankovo ;)

moorsey, 2019-06-16:
Thanks for these, huge help having these extra features!

Quick one, I used "LaserGrbl-CustomButtons-Eng-Adnan.gz"  Is there anyway of extracting the actual icon images?  Tried opening the gz file, but not a valid archive.

Essentially, I want to rotate one of the icons (below) to use for another button.  Would love them to all be the same style!
[image]

Stankovo, 2019-06-16:
Здравствуйте. Вы имеете в виду этот значок?
[image]

moorsey, 2019-06-20:
Hi @Stankovo , afraid not, can you see the one I linked to?

Thanks for the one you posted however, I could make something work with that I am sure!

Just a shame existing ones can't be extracted

Many thanks

Stankovo, 2019-06-21:
Простите, немного не понимаю вас.

jillionairedesigns, 2020-01-27:
Just started working with this program.  Is it possible to create a custom button that will move the laser to the leftmost or rightmost point that will be burned?  I have a large circular image that exceeds the size as a single file.  I would like to be able to line up the leftmost or rightmost point of the circle to burn in phases (in quarters).  Combined with the center point of the circle, I should be able to execute?

Errol1951, 2020-04-01:
I have just joined the site and the help available is fantastic thank you

arkypita, 2020-04-01:
Hi all! These are my buttons, the code I took almost everything from your contributions but I focused on giving him some rather captivating graphics.
The images are .png with transparent background.

[image]

[MyButtons.gz](https://github.com/arkypita/LaserGRBL/files/4415289/MyButtons.gz)

stan1707, 2020-04-08:
Hallo hat jemand Code zum genauen umfahren des Bildes.
Der oben mit rechteckig funktioniert bei mir nicht

distebia, 2020-04-20:
> Ciao a tutti! Questi sono i miei pulsanti, il codice che ho preso quasi tutto dai tuoi contributi, ma mi sono concentrato sul dargli una grafica piuttosto accattivante.
> Le immagini sono .png con sfondo trasparente, quindi il colore posteriore cambierà in base al tema.
> 
> [image]
> 
> [MyButtons.gz](https://github.com/arkypita/LaserGRBL/files/4415289/MyButtons.gz)
> 
> Per installare pulsanti personalizzati basta scaricare questo file, avviare LaserGRBL, fare clic con il tasto destro nell'area dei pulsanti personalizzati, selezionare "import" dal menu e aprire il file scaricato.

Ciao, potresti condividere i file .png contenuti nei vari pulsanti? ne vorrei creare altri tipo taglia immagine utilizzando quello esistente. Potresti anche implementare la funzione copia e incolla sui pulsanti. lo copio e lo modifico. Grazie

arkypita, 2020-04-20:
> Ciao, potresti condividere i file .png contenuti nei vari pulsanti?

Eccoti accontentato.
[CbIcons.zip](https://github.com/arkypita/LaserGRBL/files/4503376/CbIcons.zip)

distebia, 2020-04-20:
grazie

LoganOn1990, 2020-04-22:
Hi a scripts not work for my :( Script from this site (Turn on laser for focusing) not work in my Laser 5.5MW. 
Script button > https://imgur.com/arCImEH
And this is my Laser Module >https://imgur.com/9S3Fa3S
And front Photo > https://imgur.com/CETRiOC

Please help my :( I can't focus my laser ....

Errol1951, 2020-04-22:
I'm no expert with code but I just turn on laser with a low setting and it works for me I think I use something like s30

LoganOn1990, 2020-04-22:
> I'm no expert with code but I just turn on laser with a low setting and it works for me I think I use something like s30

Thank you i change from S5 to S30 :)

LoganOn1990, 2020-04-22:
Yes its work ! Thank you ;)

Otttomann, 2020-04-30:
I was working on a project and lost my track . I would like to set up a button, like set home button, to set center point of the image. Can someone please help me how to create this custom button?

[20 more replies]