arkypita/LaserGRBL #308: Problem with connect on my laser engraver

mitakata, 2018-03-21:
Hello, 
I have DIY laser engraver, bought from e-bay. He basset on Arduino Nano.
I have a problem when tray to connect my laser engrave to LaserGRBL. 
Driver is install properly, the laser connect and works well with BENBOX, but I can not connect to your software

This is recorded in session log:

21.3.2018 г. 17:35:10	Program     	------- LaserGRBL v2.8.39 START -------
21.3.2018 г. 17:35:14	SetStatus   	Machine status [Disconnected]
21.3.2018 г. 17:36:02	SetStatus   	Machine status [Connecting]
21.3.2018 г. 17:36:02	OpenCom     	Open COM3 @ 9600 baud
21.3.2018 г. 17:36:02	SendImmediate	Send Immediate Command [0x18]
21.3.2018 г. 17:36:12	OpenCom     	Connection timeout!
21.3.2018 г. 17:36:12	CloseCom    	Close COM3 [CORE]
21.3.2018 г. 17:36:12	SetStatus   	Machine status [Disconnected]
21.3.2018 г. 17:37:01	SetStatus   	Machine status [Connecting]
21.3.2018 г. 17:37:01	OpenCom     	Open COM3 @ 115200 baud
21.3.2018 г. 17:37:01	SendImmediate	Send Immediate Command [0x18]
21.3.2018 г. 17:37:11	OpenCom     	Connection timeout!
21.3.2018 г. 17:37:11	CloseCom    	Close COM3 [CORE]
21.3.2018 г. 17:37:11	SetStatus   	Machine status [Disconnected]
21.3.2018 г. 17:37:31	SetStatus   	Machine status [Connecting]
21.3.2018 г. 17:37:31	OpenCom     	Open COM3 @ 115200 baud
21.3.2018 г. 17:37:31	SendImmediate	Send Immediate Command [0x18]
21.3.2018 г. 17:37:41	OpenCom     	Connection timeout!
21.3.2018 г. 17:37:41	CloseCom    	Close COM3 [CORE]
21.3.2018 г. 17:37:41	SetStatus   	Machine status [Disconnected]

Sorry for my bad english and thanks in advance for hellp!

arkypita, 2018-03-21:
http://lasergrbl.com/en/faq/

**Does LaserGRBL work with my engraver?**
LaserGRBL work with any engraver that use Grbl as controller. This also include a lot of mounting kit from ebay, bangood, aliexpress, amazon, based on arduino uno/nano or other customized boards with ATmega328P chip.

Often some of these markers do not use Grbl but they use other control firmware/protocols (benbox or others) but is it possible to reprogram their firmware to use Grbl. You must google about how to do that with your specific controller.

closed 2018-03-21 by arkypita

mitakata, 2018-03-21:
Thanks for the quick response!

 >-------- Оригинално писмо --------

 >От: arkypita notifications@github.com

 >Относно: Re: [arkypita/LaserGRBL] Problem with connect on my laser engraver
 (#308)

 >До: arkypita/LaserGRBL  

 >Изпратено на: 21.03.2018 18:02

 
 
  
 http://lasergrbl.com/en/faq/   
  
 Does LaserGRBL work with my engraver? 
 LaserGRBL work with any engraver that use Grbl as controller. This also include a lot of mounting kit from ebay, bangood, aliexpress, amazon, based on arduino uno/nano or other customized boards with ATmega328P chip.  
  
Often some of these markers do not use Grbl but they use other control firmware/protocols (benbox or others) but is it possible to reprogram their firmware to use Grbl. You must google about how to do that with your specific controller.