arkypita/LaserGRBL #309: Laser dont working

Denisseb19, 2018-03-21:
Hi 
I have the NEJE machine and im trying to engrave on wood, I changed all the parameters but it still doesnt work. 
I have the last version of the program 2.8.39
When I import the image i use this parameters: 
Smooth (HQ Bicubic) 
Line to Line 
Direction: diagonal 
Quality: 7.000 lines/mm 
Previous line 

The target image 
I don't have the jogging speed only the engraving speed 

Engraving speed I use 3000
LASER ON M3 (the only option) 
LASER OFF M5

S-min 0 
S-max 255

Image size 50 X 50

I need fast engraving, so how may I change this parameters so that it works? 

Thanks

arkypita, 2018-03-24:
> The target image
> I don't have the jogging speed only the engraving speed

This is correct. I have removed the control of "jogging speed" cause the jogging speed (G0 rapid movement speed) is managed by grbl as the maximum speed configured in $$ so it was useless to take this parameter.

Denisseb19, 2018-03-25:
Ok, but I have a question, I don't know why the machine isn't working, I
don't know if are the configurations or something?

Thanks

arkypita, 2018-03-25:
please be more specific, what does really mean "isn't working"

LaserGRBL work with any engraver that use Grbl as controller. This also include a lot of mounting kit from ebay, bangood, aliexpress, amazon, based on arduino uno/nano or other customized boards with ATmega328P chip.

Often some of these markers do not use Grbl but they use other control firmware/protocols (benbox or others) but is it possible to reprogram their firmware to use Grbl. You must google about how to do that with your specific controller.

Does your engraver use GRBL as controller?

Denisseb19, 2018-03-25:
Yes, it does use it, the problem is that the machine is working, like it
moves, but i don't know why it doesnt burn the wood, like nothing happens
to the wood.

Thanks

StuartB4, 2018-03-25:
Sounds like the laser is faulty, does it burn any other materials like plastic, paper etc.
Also, if it's a Neje machine what power is the laser? if it's only a few hundred milliwatts and the material you want to burn is too light in colour it may not work. The darker the better for low power lasers.

Denisseb19, 2018-03-25:
Mmm i havent tried with other materials, let me check. 
Its neje but i dont know the vw. 
But the z values arent important to carve wood? Or i need certain values? 

Thanks!

Icemanhpg, 2018-03-26:
A setting of 3000 for a milliwatt laser is much too fast to get a decent image. Try using setting s such as 1500 or even less?

arkypita, 2018-03-26:
also note S-min 0 and S-max 255 is ok only if $31=0 and $30=255 as well

Denisseb19, 2018-03-26:
Hi
I think I got the problem
When I connect the laser machine the baud is on 9600 and it appears grbl
0.8 c.
I don't get more than $22 values.
I don't know if this is relevant?
Thanks for your help!

Icemanhpg, 2018-03-26:
I think you will have to upgrade to GRBL 1.1 to get the best results

HP Greyling

Denisseb19, 2018-03-26:
> issuecomment-376171008
> ATXa7djN3Te5Y1TadxuRGnvivJHoANMYks5tiPEAgaJpZM4S1pJU

Where do I download it?
Thanks

Denisseb19, 2018-03-26:
> nt-376207424>,

I already download the package but I can't download it, do I need to
download something else?

Thanks !

arkypita, 2018-03-26:
> When I connect the laser machine the baud is on 9600 and it appears grbl 0.8 c.
> I don't get more than $22 values.

Grbl is the software that run on-board on your NEJE that translate GCode sent by LaserGRBL in motor movement and laser pulses. It is not a software that you simply download and run on your computer: it is a firmware that is written on ATmega328P chip that is present on your electronic board.

Grbl v0.8 use 9600 as baudrate, and has only 22 config words: https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.8

Grbl 0.8 is a very old version of grbl (2013) that, for what i know, does not support power modulation (it cannot make grayscale engraving) and does not have a lot of new features. LaserGRBL can work with it, but you must disable PWM feature unchecking the first option in this window.

[image]

Disabling PWM in LaserGRBL cause that LaserGRBL generate only M3 and M5 (laser on-off) command and does not generate S0-255 power modulation.

M3 should turn on your laser, M5 should turn off. Try sending this command by hand.
If they don't work: your laser has problems. Solve it before anything else.

You can try to upgrade you grbl version, it is not so difficult as it seem (it require flashing your Atmega), but some important changes was done between 0.8 and 0.9 and 0.9 and 1.1 that involves hardware pin mapping and different and more extended configuration. If you decide to update take a copy of your $$ to know how to reconfigure grbl after update (config is lost with update).

The official site of grbl is: https://github.com/gnea/grbl where you can find more info about grbl and how to install/update it.

arkypita, 2018-03-26:
Because this hardware pin mapping changes from v0.8 and 0.9 (required for power modulation) if you upgrade to v1.1 you need to modify your hardware. You need electronic skill to do that, so i discourage you to do that if you are not an expert.

Try LaserGRBL with PWM disabled and use basics M3 M5 function of your hardware.

Icemanhpg, 2018-03-26:
If you want to upgrade your GRBL you have to flash your arduino chip.

Here is a link how to do it: https://github.com/grbl/grbl/wiki/Flashing-Grbl-to-an-Arduino.

Your pin layout of your Arduino will also change after upgrade.

Denisseb19, 2018-03-26:
> wiki/Flashing-Grbl-to-an-Arduino.

So I think its my laser machine since it isn't burning other material or
anything.
Which machine you recommend me to buy to burn wooden phone cases at home?
I also didn't understood well what to do to change that of grbl v1.1, what
is your advice?

Thanks!

arkypita, 2018-04-26:
At least 2W laser.
Grbl is born for cnc machine, v1.1 of grbl add support for [laser mode](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode)

closed 2018-04-26 by arkypita