arkypita/LaserGRBL #311: Crash on reload file during rendering

Effer, 2018-03-23:
How to reproduce:
- Open a big file (either an image or gcode).
- During bakground rendering (before the image is show in the main screen), open another file
- Crash on  DrawOnGraphics(Graphics g, Size s) method

That's seems due to GrblCommand list been modified during rendering, causing null reference.
It seems like GrblCore.OpenFile() implements a Flag CanLoadNewFile that should prevent this. But that flag is not setted properly.
Am I wrong?

arkypita, 2018-03-24:
Bug is real but the solution is not the usage of "CanLoadNewFile" since "CanLoadNewFile" is used for UI button "load file" that should be disabled when not connected or during job execution, not when redrawing preview.

The issue must be managed via try catch system in redrawing preview - aborting background drawing thread when a new file is under loading.

closed 2018-03-24 by arkypita

Effer, 2018-03-25:
Thanks for the fix and detailed description. Useful to dig into the code.

commit b31bd5d9 (2018-03-24): fix #311