arkypita/LaserGRBL #312: images leaning to one side

Icemanhpg, 2018-03-26:
I have been using LAsergrbl for a while now and after one of the previous updates I started getting an issue with that my image  " leans" to the left. After the job has completed my laser doesn't return to the zero where it started. The larger the image to be burnt the larger the leaning effect. Must also say it only happens when burning line by line or dithering. Vector burning is not affected.

To try and rectify this issue I have:
- Rebuilt the entire machine
- Adjusted the power supply to the servos
- Replaced the belt and checked tightness
- Checked wiring ..all good
- Slowed the burning speed right down to 400mm
- Changed the speed of data transfer to all possible settings.
- Replaced USB cables.

Not one of the above actions addressed my leaning effect. It is as if laser stops to early on the right of the image and too late on the left hand side of the image thereby making the image lean to the left.

Any suggestions on how to address this issue?

Thanks in advance

arkypita, 2018-03-26:
Hi @Icemanhpg 
There was a major changes in version v2.8.38 that could impact with line2line engraving. 
Up to the version v2.8.37 the line2line gcode was generated in G91 incremental mode, from version v2.8.38 onwards the line2line output was generated in G90 absolute mode.

This changes was done to get two improvements:
- G90 absolute mode is safer, because a communication error that make grbl tracing a wrong segment will not affect others next segments
- Fix a rounding error that cause cumulation of approximation remains that can cause error especially when used in unidirectional engraving mode.

You can try the older [v2.8.36](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.36) and compare with the actual [v2.8.40](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.40) to check if the difference is caused by this change.

Icemanhpg, 2018-03-26:
Thank You. Will give it try

HP Greyling

Icemanhpg, 2018-04-02:
Hi 

It doesnt matter which version I use the images still lean to the left. I see another issue has been opened with the same problem.

arkypita, 2018-04-03:
can you share a picture of the issue?

Icemanhpg, 2018-04-03:
This was a very slow burn 700mm/ min. It was a line by line  and then
vector thereafter. As per the settings of filling.

You can see that the vector that was done after the line by line us at
least 1mm of to the left.

I lnow this does not show my issue very good but as asoon as I have more
time I will send a better example.

you can clearly

Icemanhpg, 2018-04-03:
Found I have another example on my phone. If you look closely at the burnt
image you will notice that the image has a line that starts on the dark
wood. It should have started 2mm to the right of where it is. The right
side of the image has at the top a 4 mm gap between image end and dark wood
due to it leaning to the left. Should have been a 2 mm gap to the left and
right.  Further if you look at the tower in the middle you will see it is
leaning to the left. The fyrther you go from the start point the more it
gets skewed. At the end of the burn the laser is also abpyt 2 mm to the
left of 0.0 and maybe 1.5mm high .

bdring, 2018-05-04:
Maybe a long shot, but....

I cannot see your images, but I saw a similar problem on a machine a long time ago. It had to do with the direction pin logic. Try flipping bits in the Grbl $3 setting (Step direction invert)

The machine would lose one micro step per direction change. It was not noticeable on vector images where there are a tiny fraction of direction changes compared to a raster. That machine was not running Grbl or LaserGrbl, but that fixed the problem.

Icemanhpg, 2018-05-04:
Ill give it a try the moment I get to my machine. Will let you know.

closed 2018-05-31 by arkypita