arkypita/LaserGRBL #314: Picture burns out line by line

Grizz5280, 2018-03-27:
I'm not sure what I can do, but my pictures seem to show every individual line and don't look like one solid picture.  Any suggestions on how to adjust this?

[image]

arkypita, 2018-03-27:
You can:
- increase line2line resolution (line/mm)
- unfocus your laser 
- use slower speed (darken and wider burn)

Usefull tips for engraving:
http://www.instructables.com/id/10-Tips-and-Tricks-for-Laser-Engraving-and-Cutting/

mayhem2408, 2018-03-27:
@Grizz5280 and if the above fails to resolve the issue, check your stepper resolution. Make sure you are running a high microstep to allow the line to get a close as possible.

Grizz5280, 2018-03-28:
Thanks so much for the help.  I got this to work out perfectly.  I just increased the # of lines per mm.

Grizz5280, 2018-03-28:
Thanks for the help. I was able to get this to work out great by just increased the lines per mm.

closed 2018-03-28 by arkypita