arkypita/LaserGRBL #315: to do list

DJ-MOEses, 2018-03-27:
hello, is there a possibility that one can run several steps in a row?
I engrave large objects with vector and 2,000 lines diagonally. on the other hand, small phrases with vector and none, then a picture with line to line etc. the settings change with each step of the work. It would be great if you can give the program the various steps and pictures in a to do list and he works it off without a break.
ps. thanks for the great program.

arkypita, 2018-03-27:
Could be a "append file" instead of a "load new file" option good for you?

MarceloFreitas028, 2018-03-27:
Hi. I think that for a perfect engrave the laser head must move at the same speed in all engraving area so maybe in the future the deceleration and acceleration ramp can be done after and before the engraving area. The professional machines like Epilog, Gcc, Universal and Trotec do it like this.

Best Regards 
Marcelo Freitas

arkypita, 2018-03-28:
@MarceloFreitas028 
Your post is out of topic with @DJ-MOEses  request. Please open new issues for feature request.
However the acceleration offset-border is already in [development roadmap](https://github.com/arkypita/LaserGRBL/issues/64)

DJ-MOEses, 2018-03-29:
Hello and thanks for the quick reply. Adding file would work but it would also have to add the option "import raster image" for each file and "target image" should be available to change the parameters.
thank you
DJ_MOEses

arkypita, 2018-04-05:
v2.8.41 can add multiple file in a single job
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.41

closed 2018-04-10 by arkypita