arkypita/LaserGRBL #317: Want to support a grbl 1.0c laser enabled version?

Larry-M, 2018-03-29:
I have a version of grbl based on 1.0c that is being used by about 3000 people at this time. Your program will not connect to my version although it does print out the introduction information. I think that most likely the problem is a simple one.

I tested with LaserGRBL 2.6.7 and version 2.8.4.

If you would like to take a look at it you can find the distribution files here.
https://discuss.inventables.com/t/xcarve-jtech-grbl-latest-release-9-1-2017/45599

Thanks.

arkypita, 2018-03-29:
I will test it in few days

jasoroony, 2018-03-31:
I also have a couple of machines that are based on 1.0c - but also a new laser that uses the current 1.1f..  I don't think I'd want to run that machine on 1.0c anymore because of all the laser improvements made to grbl since then.  I'm not sure if the output from a job would be as good, since I think the laser is now controlled within the ISR.  I would still like to consolidate down to one gcode sender though, and LaserGRBL has been pretty good for that.  So I wouldn't mind it working on older firmwares.

I think the reporting has vastly changed since 1.0c.  I think it was fairly static back then, but now it gives you different statuses in one-off output when you send ?.

Larry-M, 2018-03-31:
@

Larry-M, 2018-03-31:
You have some mis-information concerning this issue.

The only thing you lose by using my 1.0c is the feed rate and spindle speed over rides.

My version of grbl will run your laser jobs at a faster feed rate than 1.1f which makes a difference for the higher power laser diodes (5 - 6 watts).

Another thing that my users like is that my version does not alter the way that G-code works when you are using your laser. One example is that M3 Sxxx will turn the laser on immediately, you do not have to issue a motion command to turn the laser on/off.

The laser is controlled by the ISR in both versions (my 1.0c and 1.1f).

Status reporting in 1.0c is controlled by the user, in 1.1f a status report does not return all of the information requested unless a change in status has occurred which forces GUI senders to parse the data to see what was returned and what wasn't.

You will actually get slightly better results with my 1.0c than 1.1f with your laser work. Sometimes the difference is not noticeable depending on the material you are engraving/imaging.

These reasons are why I have almost 3000 users.

But you have a choice, now that 1.1f is out, lasers are officially supported by grbl. Use what works best for you.

jasoroony, 2018-03-31:
@Larry-M 

I don't think I mentioned anything about "your 1.0c" in my post, so I do feel that it's a bit harsh to say I have mis-information concerning this issue.  For example, my 1.1f runs on a 32 bit board, and you say the version you have will run a faster feed rate than mine, even though it's already pushing the hardware to its limit now, but I wouldn't say to you that you have mis-information concerning this issue.  The M3 thing is also a safety mechanism, one that's easy to work around for focusing the laser, I wouldn't be using that as an example why users would or should be happier.  But that's just my opinion.

Thanks, and all the best.

Larry-M, 2018-04-01:
@jasoroony 

Ok, I apologize.

Since this thread was about my version I assumed that you were talking about my version.

I did make an assumption about what you were running. My mistake. I should have just said that my version will run at 199.9 IPM on Inventables  Arduino based machines and that 1.1f will not.

I appreciate your opinion as well as the opinions of the other 3000 people I deal with on this issue.

Just thought that the LaserGRBL author might like to have access to that group of people for distribution of his product.

Most of the people that run my version have issues with the "safety features" that are in 1.1f, that's why they run my version instead of 1.1f (along with the speed advantage).

BTW, my comments only apply to laser work with 1.1f. I don't have any issues with the spindle portion.

jasoroony, 2018-04-01:
No problem Larry.  I also don't have any issues with 1.0c with my mill/router, which is why I'm not going to upgrade it anytime soon and would like to see that firmware supported here.  That version would be quite popular I feel, as it was around the time many new projects appeared on the Internet.

arkypita, 2018-04-03:
@Larry-M 
I have downloaded and flashed the version Xcarve_JTech_grbl_9_1_2017.zip76 (89.9 KB)
It work with LaserGRBL as you can see in video: https://youtu.be/wJxQxejCLlA

Larry-M, 2018-04-03:
That's good news.

It doesn't work for me.

I'm running Windows 7 (32 bit). Is there some way to get some diagnostic information from LaserGRBL? Log file of some kind to see why it won't run for me? Where would a log file be?

This is what I get when I press the Connect button.

[image]

arkypita, 2018-04-03:
post your session log: ? -> open session log

arkypita, 2018-04-03:
also post a copy of your grbl configuration: Grbl -> Grbl Configuration, select all table lines, then copy and paste

Larry-M, 2018-04-03:
I don't see an option anywhere to open session log.
If I open the grbl settings location the window comes up with everything greyed out and a message that says Read and Write enabled only when connected and idle.

Here are my grbl settings using a different program.
$0=30 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=3 (dir port invert mask:00000011)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=18 (status report mask:00010010)
$11=0.005 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=25.000 (homing feed, mm/min)
$25=750.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$30=255. (rpm max)
$31=0. (rpm min)
$32=1 (laser mode, bool)
$100=40.000 (x, step/mm)
$101=40.000 (y, step/mm)
$102=188.947 (z, step/mm)
$110=7650.000 (x max rate, mm/min)
$111=7650.000 (y max rate, mm/min)
$112=2540.000 (z max rate, mm/min)
$120=8000.000 (x accel, mm/sec^2)
$121=8000.000 (y accel, mm/sec^2)
$122=150.000 (z accel, mm/sec^2)
$130=290.000 (x max travel, mm)
$131=290.000 (y max travel, mm)
$132=88.000 (z max travel, mm)

arkypita, 2018-04-03:
Session log is under menu "?"

arkypita, 2018-04-03:
Issue is related with $10=18, with $10=3 LaserGRBL work like a charm.
I think LaserGRBL made an error interpreting data sent with status report $10=18

investigating...

Larry-M, 2018-04-03:
[sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/1871767/sessionlog.txt)

Larry-M, 2018-04-03:
Yes, that's it. If I change $10=3 it works fine.

Larry-M, 2018-04-03:
Most of my users will use either $10=18 or $10=115

arkypita, 2018-04-03:
The main problem is that grbl status report format it has never been stable, they have changed it between versions and it is polymorphic due to configuration, so writing a good parse procedure is trivial. Moreover, for version 1.0 there is no documentation.

So i have to debug it and perform some "reverse engineering", i'll ask to grbl developers if there are somewhere a documentation for 1.0 status report.

Larry-M, 2018-04-03:
Will this help.

https://discuss.inventables.com/t/changes-to-the-status-report-for-grbl-version-1-0c/22434

arkypita, 2018-04-03:
check v2.8.41 https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.41

commit 33043be0 (2018-04-03): fix #317
closed 2018-04-03 by arkypita