arkypita/LaserGRBL #319: Live Path Preview

Delaser, 2018-04-01:
Hey!

Feature Request: The ability to run through an entire (Or possibly percentage of) a finished gcode with the laser power and movement speed at a custom rate. (Eg, minimal power, maximum speed)

I'd like to use this for confirming that my design will align properly and fit on my material without actually engraving it.

Thanks for hearing me out!

StuartB4, 2018-04-01:
You can use a custom button to turn on the laser at a lower power, then run the gcode, stop it when you want and use one for turning it off afterwards.

Delaser, 2018-04-01:
Can you elaborate?

From what I've experienced, running the code immediately sets the laser power as defined in the gcode. Meaning id have to create 2 versions of every run.

arkypita, 2018-04-03:
Hi, with [custom button](https://lasergrbl.com/usage/custom-buttons/) you can easily add what you asking for, in a very customized way to your hardware.

F1000
M3 S10
G0 X[left] Y[bottom]
G1 X[right]
G1 Y[top]
G1 X[left]
G1 Y[bottom]
G0 X0 Y0
M5

In custom button you can also reconfigure grbl by sending $num=val to change PWM min/max scale and set a low power mode i.e. scaling power from 0-255 to 0-2550.

closed 2018-04-03 by arkypita