arkypita/LaserGRBL #320: Laser Plotter Picture are in an Angle

Steffen88, 2018-04-01:
Hello There, i take your Great Laser GRBL,
all works fine, the quality of the Pictures are also nice, but i get one Problem:

Every Time i create a new Grafic, the Plotter print my Grafic in an Angle.
I don`t know what it`s happend.I take a lot of experimentals, but the Problem Stay.
My Workpiece are even with the Plotter!

What i can do/what`s goes wrong?

Greetings from Germany

I donate your great work!

Icemanhpg, 2018-04-02:
Steffen I have the same problem and have logged the issue as issue #312. You can see what I tried it might possibly help you.

arkypita, 2018-04-03:
@Steffen88 
can you share a picture of the issue?

Steffen88, 2018-04-04:
@icemanhpg, i'm Story, i don't Reader Your posting.

@arkypedia:yes, i upload a picture.

Greetings Steffen

Steffen88, 2018-04-04:
Hi there,
http://www.bilder-upload.eu/show.php?file=1cca75-1522843645.jpg

Ich hope the Link Workshops.

arkypita, 2018-04-04:
LaserGRBL work in a cartesian plane coordinate system, with 0,0 in the bottom left angle, x increasing to the right, y increasing going away from observer.

If you play with positioning/jogging arrows the laser movement should reflect the pressed button/arrow direction. If not, you must configure your grbl to invert axes direction.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask

Steffen88, 2018-04-04:
Thank you very much!

closed 2018-04-12 by arkypita