arkypita/LaserGRBL #321: Issue detector	UnexpectedReset [2,-1,-1] - Arduino UNO clone with CH340G - constantly disconnects from the machine

dbConstantine, 2018-04-02:
Hi all,
I've recently built a 2 axis laser engraving machine controlled by 3 easy driver 4.4 stepper drivers (2 motors for Y axis ) and one cheap Arduino clone with a CH340G chip for COM. 
I've wrote grbl 1.1f to the board and I'm using LaserGRBL to stream data to the machine however, midway through projects, the interface shows the message "LaserGRBL has detected an unexpected error with your board" and cuts off. 

The error is consistent regardless of the project so the only usefull piece of information I can provide without guidance is the below sample of the session log.

Is this a known / common issue I can address with the hardware listed above? I'm asking since I've read about similar issues with the CH340 chips and I'm wondering if I need to change to a board with ATmega16U2 ? 

Log sample:

4/2/2018 9:44:14 PM	Program     	------- LaserGRBL v2.8.40 START -------
4/2/2018 9:44:17 PM	SetStatus   	Machine status [Disconnected]
4/2/2018 9:44:29 PM	SetStatus   	Machine status [Connecting]
4/2/2018 9:44:29 PM	OpenCom     	Open COM3 @ 115200 baud
4/2/2018 9:44:29 PM	SendImmediate	Send Immediate Command [0x18]
4/2/2018 9:44:29 PM	VersionInfo 	Detected Grbl v1.1f
4/2/2018 9:44:29 PM	SetStatus   	Machine status [Idle]
4/2/2018 9:44:38 PM	OpenFile    	Open C:\Users\username\Desktop\image.png
4/2/2018 9:44:53 PM	EnqueueProgram	Running program, 7482 lines
4/2/2018 9:44:54 PM	SetStatus   	Machine status [Run]
4/2/2018 9:50:57 PM	VersionInfo 	Detected Grbl v1.1f
4/2/2018 9:50:57 PM	Issue detector	UnexpectedReset [2,-1,-1]
4/2/2018 9:51:01 PM	SetStatus   	Machine status [Idle]

arkypita, 2018-04-03:
The UnexpectedReset evidence that your grbl on arduino perform a self reset, sending an unexpected "Welcome Message" (`Grbl X.Xx ['$' for help]`) that LaserGRBL receive and signal to you with the issue detector.

The reset of grbl/arduino evidence that your hardware is faulty or misconfigured.
Please have a look at http://lasergrbl.com/faq to check common causes of this issue.

I am using CH340 clone, with success, but in my experience with clones I have found working ones and totally faulting ones. At faq page you can find some hint about how to configure LaserGRBL and Grbl to be less faulty (more error tollerant, less hardware stressing).

If nothing work, consider change with an original arduino or a clone with ATmega16U2

dbConstantine, 2018-04-03:
Hi,

Thank you for the reply and for pointing out that it is indeed a hardware issue. 

I'll review the machine for any faults and/or misconfigurations as well as any optimizations I can make to it based on the FAQ page. 

If I do find a specific cause for the error I'm experiencing, if ok with you, I'll post it here along with a more detailed description of the hardware in case anyone else needs this information.

arkypita, 2018-04-06:
> In my case, the issue was with the firmware in conjuncture with the board I'm using so unrelated to LaserGRBL.
> 
> Grbl 1.1f takes up a significant part of the board's memory (~93%) on a 328p chip with 32kB and this leads to instability in operation.

I use Grbl 1.1f with 328p and it work. Maybe your is faulty?
Please note that grbl cannot manage too high step frequency, what is your $100, $101, $102 configuration? https://github.com/gnea/grbl/issues/249

dbConstantine, 2018-04-06:
the values for $100, $101 and $102 on my machine are 42.223 step/mm.

closed 2018-04-12 by arkypita

Revenger72, 2018-05-29:
A similar problem has arisen. Maybe because of the speed of 4000-6000?
After some time stably stops the machine and burns the laser. All coordinates are reset.

Photo: https://2.downloader.disk.yandex.ru/preview/4bf6a885f32923b568cedb6d81b26e1d467973f8b34f561e7d589e98b774a498/inf/RBpp162fE4IHHoO3piGj8Nkhkf4NbRqTen2ZYXMlQ8Qz5in2Z-iW5okWuQX9-9mIiCc9HoHj8bCHZuDtvjyP1g%3D%3D?uid=5163375&filename=2018-05-29%2014-08-12.JPG&disposition=inline&hash=&limit=0&content_type=image%2Fjpeg&tknv=v2&size=1163x716

arkypita, 2018-05-30:
Speed of 4000- 6000 is not a problem unless $100, $101 $102 use low value.

Revenger72, 2018-05-30:
Then I do not know the reason. One day worked perfectly, and the second started ...
  The number of steps is set to 40 (X) and 80(Y). I think this is quite standard for the motors NEMA17