arkypita/LaserGRBL #322: home

i3hvs, 2018-04-05:
Buon giorno mi scuso se non scrivo in inglese ma purtroppo non lo conosco.
Sono passato da Win XP a Win10 e qui è sorto il problema.
Sto utilizzando il progetto di CNC laser 3bpBurner2st con il relativo GRBL 0.9 dell'autore mentre per l'esecuzione utilizzo LaserGRBL sia per il dithering dell'immagine sia per l'esecuzione quindi mentre utilizzavo XP non riscontravo alcun problema, la macchina si portava in posizione di home e iniziava la scrittura lavorando normalmente ora con Win10 la macchina, dopo aver effettuato l'home regolare, in basso a sinistra, parte portandosi tutta a destra fino al fine corsa ed insistendo per portarsi oltre.
E' capitato ad altri questo problema? e come avete risolto?
Vi ringrazio fin da ora per l'eventuale aiuto
Vittorio i3hvs

arkypita, 2018-04-05:
Premesso che il codice generato da LaserGRBL non ha nulla a che fare e non dipende dalla versione del sistema operativo (quindi non trovo un nesso) sai dire che versione di LaserGRBL stai usando su W10 e che versione usavi su XP?

Hai ancora il pc con XP? Puoi provare a generare un semplice gcode su XP e su W10 usando la stessa immagine e le stesse impostazioni e confrontarli? (file->salva programma, poi li apri con editor di testo)

i3hvs, 2018-04-05:
Ti ringrazio per la pronta risposta la prova che tu dici l'ho fatta utilizzando la stessa immagine sia su XP e sia sul win10, ho ancora l'altro computer con XP, è che mentre il gcode generato da win XP funziona regolarmente quello generato da win10 presenta quell'inconveniente.Su XP uso la versione .38 ma ho provato a installare anche la .40 e va bene lo stesso mentre sul 10 ho provato tutte le versioni ad iniziare dalla .40,.42,.38,.39 ottenendo sempre lo stesso risultato di "sbattere" sul estremità destra dell'asse.
ps. il file generato, sempre con gli stessi parametri, da win 10 è pari a 4.555 Kb mentre su XP è di 5.666 ma penso che ciò sia dovuto ai miglioramenti 
Spero di essere stato esauriente

arkypita, 2018-04-05:
se puoi zippare due file, uno generato su w10 e uno generato su XP (stessa versione di LaserGRBL, stessa immagine, e stesse impostazioni di conversione raster) e caricarli qui, posso provare a darci un occhiata

i3hvs, 2018-04-05:
[prova1.zip](https://github.com/arkypita/LaserGRBL/files/1879172/prova1.zip)
[prova3.zip](https://github.com/arkypita/LaserGRBL/files/1879173/prova3.zip)
[immagine.zip](https://github.com/arkypita/LaserGRBL/files/1879174/immagine.zip)
prova 1 generata da win10-
prova 3 generato da XP
immagine è l'immagine di partenza 
grazie della disponibilità

arkypita, 2018-04-05:
Sei sicuro di star usando la stessa versione di LaserGRBL su entrambi i PC?
Così a occhio direi che su XP hai una versione 2.8.36 o più vecchia

i3hvs, 2018-04-05:
di prove ne ho fatte diverse è facile che abbia mischiato un po' i file rimane però il fatto che il problema rimane tra i due sistemi operativi pazienza vorrà dire che adopero l'XP per generare i codice g e poi il 10 per stampare. Per adesso ti ringrazio resta in ogni modo il tuo un ottimo programma complimenti per lo sviluppo

arkypita, 2018-04-05:
Ripeto che non c'entra la differenza tra sistemi operativi: stai usando due versioni differenti di LaserGRBL, punto.

Dalla versione 2.8.38 in poi viene utilizzata la modalità G90 (movimento assoluto) per il codice generato, mentre nelle versioni più vecchie si usava la modalità G91 (movimento incrementale/relativo).

Si vede chiaramente dalle prime 10 righe di gcode:

Il file generato su XP con versione vecchia di LaserGRBL:
```
G90
F4000
G0 X0 Y0
M3 S0
G1 F1000
G91
G0 X0.2 Y0 F4000 S0
G1 F1000 X.3 S255
G0 X0.1 Y0 F4000 S0
G1 F1000 X.2 S255
```

Il file generato su w10 con una versione nuova di LaserGRBL:
```
G90
G0 X0 Y0
M3 S0
F1000
G0 X0.2 Y0 S0
G1 X0.5 S255
G0 X0.6 Y0 S0
G1 X0.7 S255
G0 X0.8 Y0 S0
G1 X1.1 S255
```

i3hvs, 2018-04-06:
Scusami se ieri non ho potuto risponderti subito ma impegni mi hanno allontanato per il resto della giornata.
Questa mattina ho ripreso in mano l'argomento, ti premetto che di codice g non ne capisco niente come pure di programmazione, sono rimasto ai tempi delle valvole.
Per la particolarità del codice scritto per 3bp Burner aggiungevo in testa i seguenti comandi:
$3=2
$H
$3=0
questo mi permetteva di portare home in basso a sinistra anziché in alto (come da programma fatto dall'autore) ora aggiungendo queste righe nelle nuove versioni , ho provato sia la 40 che la 41, succede che dopo aver effettuato l'home il carrello parte per posizionarsi tutto a destra, ho eliminato le righe da me aggiunte ed ora la macchina non esegue l'home e il laser parte esatto mi resta quindi il capire come far partire la scrittura dal punto X0 Y0.
Ulteriore prova che ho fatto è stata quella di inserire manualmente le mie tre righe e poi dare il comando per eseguire il codice il risultato è stato quello di mandare tutto a destra il carrello arrivati a questo punto penso ci sia qualche incompatibilità di settaggio nel grbl 0.9

arkypita, 2018-04-06:
LaserGRBL lavora in un sistema di coordinate del piano cartesiano, con 0,0 nell'angolo in basso a sinistra, X crescente a destra, e Y crescente in allontanamento dall'osservatore.

Se giochi con le frecce di posizionamento il movimento del laser dovrebbe riflettere la direzione del tasto -freccia premuto. In caso contrario, è necessario configurare il proprio grbl per invertire la direzione degli assi.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#3--direction-port-invert-mask

Una volta che hai configurato $3 e ottenuto questo comportamento (X crescente a destra, Y crescente allontanandosi) puoi provare l'homing.

Se l'homing non funziona più (va al contrario) puoi riconfigurare l'homing di conseguenza

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#23---homing-dir-invert-mask

i3hvs, 2018-04-06:
allora con le freccie lo spostamento è corretto mentre con l'homing l'asse y è contrario ora provo a configurare $23=3 dovrebbe funzionare ,il $23 è già a 1.
Ora home è giusto basso a sinistra però quando mando in esecuzione il carrello si porta in alto a destra

i3hvs, 2018-04-06:
penso che basti invertire gli assi con $3

i3hvs, 2018-04-06:
provato con esito negativo

arkypita, 2018-04-06:
quando torno a casa vedo come è la mia configurazione

i3hvs, 2018-04-06:
ok

i3hvs, 2018-04-06:
mi è venuto un dubbio non vorrei che sbagliassi io è giusto che l'home avvenga in basso a sinistra e la scrittura inizi in alto a destra oppure come vorrei io l'home in basso sinistra e la scrittura sempre in basso sinistra?

arkypita, 2018-04-06:
Il gcode nasce per le CNC. Le CNC essendo macchine "a togliere" hanno delle convenzioni riguardanti il sistema di coordinate un po' particolari. Un laser invece è come un plotter e per LaserGRBL ho scelto quello che è il sistema di coordinate più spontaneo, cioè quello del piano cartesiano, quindi LaserGRBL produce il gcode e disegna le anteprime secondo questa convenzione.

- switch di homing va posizionato in basso a sx
- dopo l'homing la posizione riportata deve essere X0 Y0
- la x cresce verso destra, quindi G1 X10 F100 sposta a destra di 10mm
- la y cresce allontanandosi dall'osservatore, quindi  G1 Y10 F100 allontana di 10mm

Il seguente codice fa l'homing, si sposta a X10 Y10 quindi disegna un quadratino di 10mm di lato sui punti X20 Y10, X10 Y20, X10 Y20, X10 Y10

```
$H
G90
F100
G1 X10 Y10
M3 S255
G1 X20 Y10
G1 X20 Y20
G1 X10 Y20
G1 X10 Y10
M5
G0 X0 Y0
```
[image]

Devi configurare il tuo sistema affinché si comporti così, usando i parametri $$ di grbl

Qui alcune indicazioni, ma non chiedermi di tradurtele:

https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#homing-cycle-isnt-working-right-the-movements-are-all-going-in-wrong-directions

https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to

i3hvs, 2018-04-06:
Ti ringrazio per le delucidazioni che confermano la mia teoria vedrò ora di scoprire perché l'home è giusto mentre l'inizio lavoro è sballato un'ultima cortesia che ti chiedo mi passeresti la tua di configurazione in modo da poter fare un paragone corretto? 
grazie

i3hvs, 2018-04-06:
ps poi prometto di lasciarti tranquillo

closed 2018-05-31 by arkypita