arkypita/LaserGRBL #324: Soft never connect

giuliano2505, 2018-04-08:
Well I think the problem is in my PC, but I cant find where.
When I try to connect to the machine, the soft keeps in "Connecting". Soft and machine work perfect in other pc.

The log:
8/4/2018 6:23:10 p. m.	SetStatus   	Machine status [Disconnected]
8/4/2018 6:29:27 p. m.	SetStatus   	Machine status [Connecting]
8/4/2018 6:29:27 p. m.	OpenCom     	Open COM3 @ 115200 baud
8/4/2018 6:29:27 p. m.	SendImmediate	Send Immediate Command [0x18]
8/4/2018 6:29:37 p. m.	OpenCom     	Connection timeout!
8/4/2018 6:29:37 p. m.	CloseCom    	Close COM3 [CORE]
8/4/2018 6:29:37 p. m.	SetStatus   	Machine status [Disconnected]

It's look like the machine never response after reset. But if I use TeraTerm the machine works fine.
My GRBL version is:
[VER:1.1f.20170801:]
[OPT:V,15,128]
And my $10=3

I cant find any solution for now.

arkypita, 2018-04-09:
Yes you are right, it's look like the machine never response.
With TeraTerm it respond on the same machine? Very strange...

Does the COM port disappear from LaserGRBL list when your detach usb cable? Does it appear when your reconnect it?
Did you have tried with others USB ports?

I can give you a debug version of LaserGRBL that records all the com-port traffic, if you want to have more diagnostic.

Ciao, Diego

giuliano2505, 2018-04-09:
I download it from other similar issue. comlog.txt give me this:
00008861	00000	com	Open COM10 @ 115200 baud
00008934	00001	tx	[0x18]
00008945	00002	tx	[0x3F]
.......
00018867	00052	com	Close COM10 [CORE]

I think it never response to "?", buuuuuut, if I reset by hardware the machine while "connecting", the soft connects and runs fine. And the machine send this:
00007253	00015	rx	88888888<88888888<8<888888<8<88888888<88888888<8888....
After a looooooooong line....
00007309	00016	rx	Grbl 1.1f ['$' for help]\r
And it's works automagically.

I think maybe Its a problem with the buffer, could suggest to clear it after open the port, then send the reset, and then wait for the first character ("G") and discart all before it.

TY for the quick response.

arkypita, 2018-04-10:
Try this version: https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.43
Configure for hard reset on connect. Let me know

giuliano2505, 2018-04-11:
It looks like works well, I am going to continue with test but It works.
Another thing I find is that if I connect the machine when the soft is open it connect well. I can't try it before because the cable is not very accesible, but is info.
I think this can solve the problem, so thanks!
Giuliano

closed 2018-04-12 by arkypita