arkypita/LaserGRBL #327: engraving speed

renadan, 2018-04-12:
Hi I have a problem - I can not adjust engraving speed. Very slow and burns wood. my product GRBL 0.9J, USB port cnc engraving machine control board, 3 axis control - 2.5W 450nm Blue Laser Module TTL 12V - Nema 17 engine.
Thanks for help

arkypita, 2018-04-13:
Maximum speeds are configured in grbl
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

renadan, 2018-04-13:
you still do not know which point to change. help thank you

arkypita, 2018-04-16:
The setting is stored in your engraver board.
You can access it using LaserGRBL, Menu "Grbl", "Grbl Configuration"

[image]

closed 2018-04-16 by arkypita