arkypita/LaserGRBL #329: Issues with higher COM port numbers

BallscrewBob, 2018-04-12:
It starts around the COM 25 mark.

Win 7 x65 pro decent machine spec and other GRBL senders dont seem to mind the higher COM ports.
Please don't ask me to remove unused COM ports I test for Arduino CREATE so there is always a plethora of live boards (minimum 14 for Arduino themselves and a few use two COM ports each such as the YUN and MKR series etc.).

I had an issue where I disconnected from COM 27 and closed out the program.
Restarted the program and COM 27 was still listed as the connection so all good so far.
However when trying to connect I get a message "The port 'COM2' does not exist." However it will connect to any port below 25.
Oddly it had primarily connected to that port and I had used it to do some laser tests which makes it all the more peculiar.

arkypita, 2018-04-13:
I have a piece of code that handle a known issue with bluetooth serial bridge #31 that require to try open the port with the provided COM number, and if fail try to open the same port without the last digit.
I think this is why of the error COM2 does not exist when opening COM27

As the first open is error-catched the second open is not error-catched, this is why you have the message. Imho the first open of the port with COM27 goes wrong (maybe already opened?) so you have the error on the second.

See your session log for more detailed info (menu ? - open session log)

```
try
{
	Logger.LogMessage("OpenCom", "Open {0} @ {1} baud {2}", com.PortName.ToUpper(), com.BaudRate, ResetDiagnosticString());

	com.Open();
	com.DiscardOutBuffer();
	com.DiscardInBuffer();
}
catch (System.IO.IOException ioex)
{
	if (char.IsDigit(mPortName[mPortName.Length - 1]) && char.IsDigit(mPortName[mPortName.Length - 2])) //two digits port like COM23
	{
		//FIX https://github.com/arkypita/LaserGRBL/issues/31
		com.PortName = mPortName.Substring(0, mPortName.Length - 1); //remove last digit and try again
		Logger.LogMessage("OpenCom", "Retry open {0} @ {1} baud", com.PortName.ToUpper(), com.BaudRate);

		com.Open();
	}
	else
	{
		throw ioex;
	}
}
```

BallscrewBob, 2018-04-13:
A session log will have to wait until it happens again.
As part of my debug with Arduino I use CCLEANER and WISE registry cleaner to clear cache etc.
I had run them to see if it helped which it did btw !

I doubt it was browser cache but something in a temporary file or registry key  maybe ?

arkypita, 2018-04-16:
This confirm what I say but unlucky does not give more information because the error reported is related to the second try (Retry open COM2) and the error that cause the impossibility to open the COM27 was not reported :-(

I will write a better diagnostic for this issue in the next version asap

arkypita, 2018-04-26:
the last version come with a diagnostic that record the original exception, so we can see why did not connect on COM27

BallscrewBob, 2018-04-26:
Thanks...am running latest pretty heavily with two instances and not seen it yet.

BallscrewBob, 2018-05-02:
I think this can be fully closed as I have only seen it once since the upgrade and that was probably my fault.

Thanks for your hard work !

closed 2018-05-02 by arkypita