arkypita/LaserGRBL #331: Abort during processing and FAN Control

Xillius178, 2018-04-14:
First of all, many, many thanks for such a great software !!!

In addition, freeware and in several languages, you have my full respect!

When I run a gcode program,
I can only stop it completely via the "lightning symbol".

While it may stop and continue, it does not stop.

And when I do it over "Reset Grbl" I lose my home position.

Would it be possible to implement a cancel button?

The second topic, I have my fan (for laser cooling and component) connected via my board, on the Gcodes M08 and M09 I can start and stop this.

With my Gcode files that works fine too.

But if I use the integrated image to laser option,
I have the command (have a button created) always active even before beginning.

Is it possible to add an option so that the M08 / M09 is immediately implemented?

Sometimes I forget to activate the fan beforehand, then I can only reset and lose the home position as described above.

Is it also possible that you write me an email, have a few small Inovation / ideas for the laser plotter, would like to introduce you.

Greetings, Xillius
[image]

arkypita, 2018-04-16:
> First of all, many, many thanks for such a great software !!!
> In addition, freeware and in several languages, you have my full respect!

Thank you. Translation is work of some volunteers 

> When I run a gcode program, I can only stop it completely via the "lightning symbol".
> While it may stop and continue (with the buttons on the left side), but can not abort and when I do it over "Reset Grbl" I lose my home position. Would it be possible to implement a cancel button?

Yes, it is possible and it could be easy to implement. I will add it asap

> The second topic, I have my fan (for laser cooling and component) connected via my board, on the Gcodes M08 and M09 I can start and stop this. Is it possible to add an option so that the M08 / M09 is immediately implemented?

I choose to not add a lot of options for the code generator because
1. more option means more complex UI and program and LaserGRBL want to be simple to use
2. anyone want / ask for different personalization, sometime in conflict between them, so it is not possible to satisfy everyone

but: M8-M9 control is a reasonable request, so in future i have plans to implement it.

I have in program a big code refactoring to make raster import more customizable, i will insert this changes in this task.

arkypita, 2018-04-16:
please note that both request was already on development roadmap #64

closed 2018-04-16 by arkypita

arkypita, 2019-11-23:
You can test this version: https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.19
Maybe custom header/footer can be a solution for you.