arkypita/LaserGRBL #333: translation of the program into another language (Russian localization)

Arhimed-Ru, 2018-04-16:
I tried to translate the program into Russian using my experience
here is the result:
[LaserGRBL2842_ru.zip](https://github.com/arkypita/LaserGRBL/files/1915171/LaserGRBL2842_ru.zip)

I hope this experience will help correctly translate the next version.

Arhimed-Ru, 2018-04-18:
BtnZeroing...
What is it?

arkypita, 2018-04-18:
[image]

Arhimed-Ru, 2018-04-18:
Transition to the zero point of coordinates - :)

Arhimed-Ru, 2018-04-18:
I corrected the full translation:
[LaserGRBL_ru_18.04.2018.zip](https://github.com/arkypita/LaserGRBL/files/1922938/LaserGRBL_ru_18.04.2018.zip)

arkypita, 2018-04-18:
Posting compiled exe or dll is completely useless as because on the first LaserGRBL update they will be not updated and become incompatible. If you want to do a useful work for the community you have to submit a pull request for your modified .ru.resx files so they can be included in the source.

Please note that I have suspended task of translation because i want to made some code/ui refactoring that will change the code so merging users contribution make my work more complex.

arkypita, 2018-04-18:
https://github.com/arkypita/LaserGRBL/issues/28

Arhimed-Ru, 2018-04-18:
please!!!!!   update the resource files (add phrases):
HotkeyManagerForm.resx
IssueDetectorForm.resx
NewVersionForm.resx
ResumeJobForm.resx
RunFromPositionForm.resx

Arhimed-Ru, 2018-04-18:
Files with available phrases prepared, corrected and packed into the archive:
https://drive.google.com/open?id=19nOu4224ywK9vjLX9URY119YOpLDD7Ql

Why do not you allow all phrases to be translated?
For example, here is this file:
setting_codes.v1.1.hcv
error_codes.csv
alarm_codes.csv
.....
And I can not find another ten phrases that are not in the resources...

Arhimed-Ru, 2018-04-18:
I wrote down the phrases in a table so you can see:
https://drive.google.com/open?id=1JipHKSAd2H-RaOrEkqueYhs-jRkGuqJk

arkypita, 2018-04-19:
Visual studio 2017 community, its free and can be downloaded from official site

arkypita, 2018-04-19:
> please!!!!! update the resource files (add phrases):
> HotkeyManagerForm.resx
> IssueDetectorForm.resx
> NewVersionForm.resx
> ResumeJobForm.resx
> RunFromPositionForm.resx

Yes, you are right. All those form ar recently added and I have not provided the localized resource file.
It's something I shoud do BUT as I have already say "I have suspended task of translation because i want to made some code/ui refactoring that will change the code so merging users contribution make my work more complex"

Arhimed-Ru, 2018-04-19:
Visual studio 2017 community full - too big
....
please excuse me
what is the minimum package for your program?

arkypita, 2018-04-19:
".NET  Desktop Development" is the only package needed
in alternative you can use [sharpdevelop](http://www.icsharpcode.net/OpenSource/SD/Download/Default.aspx)

Arhimed-Ru, 2018-04-19:
I installed Visual studio and a set of NET
opened the project...compiled the program...launched
the program is only English - when switching the language the program (2.8.43) is still English
if you replace the file  LaserGRBL.resources.dll  from the older version - working
the file LaserGRBL.resources.dll 2.8.43 - no  working
I can not check files *.ru.resx 
:(

Arhimed-Ru, 2018-04-19:
I checked the operation of the regulators 
Power (0.1...2.0), Linear (0.1...2.0), Rapid (0.25;0.5;1.0).
Linear and Rapid - work correctly speed changes
Power only diappazone 0.1 ... 1.0 - correctly
diappazone 1.1 ... 2.0 - It does not regulate

in my translation it is necessary to swap phrases:
"Холостая скорость"     ↔    "Мощность  лазера"

Arhimed-Ru, 2018-04-26:
I translated version 2.8.44 using "Radialix"
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

4arus, 2018-04-27:
Добрый день. Вы можете скинуть ссылку на переведенную программу.

4arus, 2018-04-27:
Ваша работа очень хорошая.

Arhimed-Ru, 2018-04-28:
4arus
даю ссылку на русский экзешник (он в архиве):
https://drive.google.com/open?id=1qU8pnHOpFZcoynAZUEqti5O344Dap515
файл извлеките из архива (можете проверить на вирусы - я сам ничего там не добавлял),
скопируйте в папку с программой (можете убрать приставку "ру")
переключать на другие языки нет смысла (этот файл с русскими фразами).

4arus, 2018-04-29:
Спасибо. Я тоже когда то пытался перевести первые выпуски. Но Visual studio выдавало ошибку, пробовал "радиаликсом" но так и не освоил эту программу. Сейчас занимаюсь автономным контроллером для лазерного станка с grbl. Как закончу поделюсь.

4arus, 2018-07-04:
Доброго времени суток. Понравился ваш перевод. А вы случаем не переводили последний релиз программы?

closed 2018-07-05 by arkypita

Arhimed-Ru, 2018-07-05:
[image]
https://drive.google.com/open?id=1IvdAXS6rn9QBXKK8iVx1-kfm_DzMC766

Mikalay31, 2018-07-22:
Не удалось скачать текстовую инструкцию.