arkypita/LaserGRBL #334: control the position along the Z axis

Arhimed-Ru, 2018-04-17:
I plan to use the Z-axis motor to adjust the focus of the laser and for the possibility of engraving on non-planar surfaces.
Please add the ability to control the position along the Z axis

arkypita, 2018-04-17:
[As documented](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#extended-ascii-realtime-command-descriptions) grbl support only 25-50-100% as rapid (M0) override so there is no way for lasergrbl to implement a higher division step.

Arhimed-Ru, 2018-04-18:
Yes. I understand....
As before, I will have to save the file in *.nc format, then edit it and load it for work....

arkypita, 2018-04-18:
> I plan to use the Z-axis motor to adjust the focus of the laser and for the possibility of engraving on non-planar surfaces. Please add the ability to control the position along the Z axis.

I have no current and no future plans to make LaserGRBL able to work on non-planar surface. Basic Z control for focus adjust can be added via custom button.