arkypita/LaserGRBL #336: Question for Italian folks :-) (maybe offtopic)

mibiosa, 2018-04-18:
Ciao Ragazzi e ciao arkypita... domanda al quale spero di avere una risposta:
Dove posso trovare in italia un negozio online che venda il controller eleksmaker mana SE?
Sino ad ora ho trovato solo Bangood o direttamente il negozio eleksmaker ma i tempi di attesa sono minimo dai 10 giorni ai 20 giorni (perché spediti direttamente dalla Cina) e mi sto disperando perché devo concludere un lavoro e il mio arduino è andato a farsi benedire!!!
Qualcuno può darmi una dritta? O qualcuno ne ha una funzionante in più da vendermi?
Grazie mille in anticipo :-)
-Miko

StuartB4, 2018-04-19:
@Arkypita. The one you point to on EBay is still from China. The seller may be in Italy but looks like he has them sent from China.
The actual address is given as:
"Informazioni sul venditore professionale
SHENZHEN SHI NAI KE SE SI Trading LTD
BAOGUANG HU
24 Floor, Wing Shing Industrial Building,
26 Ng Fong Street, San Po Kong, Kowloon
Hong Kong"

arkypita, 2018-04-19:
@StuartB4 really?

[image]

mibiosa, 2018-04-19:
@arkypita quello che mi domando anche io e che mi fa insospettire è che, visto che vengono spediti da Firenze, perché ci impiegano tanto tempo ad arrivare?
Ho notato che molti venditori cinesi mettono come sede di spedizione l'Italia ma poi, dopo aver fatto l'acquisto, scopri che l'oggetto viene spedito dalla cina :-(

arkypita, 2018-04-19:
Sono lunghi se consideriamo la spedizione da Firenze ma se venisse dalla Cina sarebbero invece troppo corti. Quando spediscono dalla Cina mettono in conto almeno 30-40 giorni.

Comunque, se devi sostituire solo l'arduino nano o i driver li trovi su EBM store (sito o ebay)

http://www.ebmstore.it

https://www.ebay.it/itm/ARDUINO-NANO-V3-0-con-cavo-USB-ATmega328-COMPATIBILE-CH340-ART-CV02/261860734061?hash=item3cf81db46d:g:uvAAAOSw44BYC0Dk

http://ebmstore.it/index.php?route=product/search&search=DRIVER

StuartB4, 2018-04-19:
@arkypita. Yes really.

A lot of ebay traders sell items from their home address but when they recieve the payment they then order them from china and have them delivered direct from china to the buyer.
Thats also why the delivery date in the listing is such a long time. It has to come from china.
If you were in italy and the item was in italy surely it would only take a couple of days.

mibiosa, 2018-04-25:
@arkypita Ciao :-)
MEssaggio solo per informarti che è arrivato l'arduino che ho comprato sul sito ebmstore che mi hai suggerito e adesso funziona tutto perfettamente.
Ti ringrazio ancora tanto :-)
Ti volevo chiedere una piccola info: esiste online un manuale completo su LaserGRBL? Hai scritto qualche cosa a riguardo? Ho visto i video sul sito ma mi sfuggono alcune cose... apro un altro topic? Magari in inglese?
Grazie in anticipo.
Michelangelo

arkypita, 2018-04-26:
http://lasergrbl.com/it/usage/

closed 2018-04-26 by arkypita