arkypita/LaserGRBL #339: Привет всем.

Koctetcav, 2018-04-23:


Koctetcav, 2018-04-23:
Все разобрался. Только при движении он работает. А я хотел узнать как при простое  лазер работал.
Чтоб края можно было отрисовывать... Можете с такой бедой помочь...

Arhimed-Ru, 2018-04-23:
Я ЖЕ ТЕБЕ НАПИСАЛ НА ФОРУМЕ - ОТКЛЮЧИ ЛАЗЕРНЫЙ РЕЖИМ - ТОГДА ПРИ ПРОСТОЕ
ЛАЗЕР БУДЕТ СВЕТИТЬ.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Без
вирусов. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Koctetcav, 2018-04-23:
Всё. Спасибо большое. Разобрался........ Огромное спасибо......

Arhimed-Ru, 2018-04-23:
по поводу других кнопок.
Кнопка "обнулить положение" (которая 00.00) - там надо исправить код 92 на 92.1 (будет по другому "сбрасывать координаты"):
G92.1 X0 Y0 Z0
тогда она не будет дублировать уже существующую кнопку, которая дает команду G92X0Y0 

Отключить Лазерный режим $32=0  - добавить в описание: "для прожига точками"
Включить Лазерный режим $32=1 (гаснет при остановке)
Включить лазер M3 (если 32=0)
.....

и перевести эту часть кнопок (где не надо загруженный рисунок) чтобы включались при соединении сразу

Koctetcav, 2018-04-23:
Ок. Я разобрался. Спасибо большое.

Koctetcav, 2018-04-23:
Подстроил кнопку обнулить положения. Все работает.

Koctetcav, 2018-04-23:
Спасибо огромное. Теперь еще больше стала нравится эта программа.

Koctetcav, 2018-04-23:
А сетку не кто не скидывал? Калибровочную...

Koctetcav, 2018-04-23:
Я себе еще кнопку сделал чтоб лазер на всю мощность работал. Чтоб отрисовывать края. Чтоб потом на ЧПУ гравере вырезать. Все получилось. Кому надо могу скинуть.

Maxmgl, 2018-04-24:
Ребята выручайте, у меня вот такой гравер https://ru.aliexpress.com/item/40X50cm-Laser-Engraving-Machine-5500mW-DIY-Laser-Engraver-IC-Marking-Printer-Carving/32801826899.html. LaserGRBL посылает команды которые он не понимает. Например если в командах программы laserGRBL ввести команду включения лазера "M3 G1 F4000 S1000" то лазер включается на полную мощь, и также если загрузить nc файл, сделанный штатной программой, которая была в комплекте - LaserGBRL работает, НО при запуске nc файла сделанного программой Lasergrbl посылается команда M3 для прожига, лазер при это не работает а только перемещается. Что это может быть?

Koctetcav, 2018-04-24:
Попробуй M4 поставь. Какая ардуино стоит? И какая прошивка?

Maxmgl, 2018-04-25:
Плата как я понял ардуино нано https://ae01.alicdn.com/kf/HTB1PFRXQpXXXXXOXVXXq6xXFXXXR/40x50-2500-5500-DIY.jpg
прошивка в командной строке lazergrbl указана 1.1

M4 попробую сегодня

Arhimed-Ru, 2018-04-25:
Maxmgl 
напиши (скопируй текстом) часть кода, который хорошо работает на твоей машине и часть кода, который не включает лазер... чтобы сравнить.

Maxmgl, 2018-04-25:
Если ничего не перепутал (станок не под руками) - вот фрагмент кода который работает(сформирован штатной программой)
;TYPE2 LINE28132 X92.9250 x0.0000 Y89.9250 y0.0000 Z0 z0
;Export By diy8000.taobao.com
S1000
G0 F2000
G92 X0 Y0 Z0
G21
G90
G0 F2000
M05

G0 X26.8500 Y3.0750
M3
G1 X68.7750
G1 Y3.1500
G1 X26.8500

А вот фрагмент, который сформирован LaserGRBL и он не работает ни в ней ни в штатной
G90
G0 X0 Y0
M5 S255
F1000
G0 X2.8 Y43.4 M5
G1 Y43.8 M4
G0 X3 Y44 M5
G1 Y43.8 M4
G0 X3 Y43.2 M5
G1 Y43 M4
G0 X3.2 Y42.8 M5
G1 Y43 M4
G0 X3.2 Y44 M5
G1 Y44.2 M4
G0 X3.4 Y44.4 M5
G1 Y44.2 M4
G0 X3.6 Y42.6 M5

Arhimed-Ru, 2018-04-25:
Maxmgl
вы должны еще и понимать разницу между включенным лазерным режимом и выключенным.
фрагмент, который не работает - сформирован в LaserGRBL как? в какой версии?
Машина была подключена? версия GRBL машины "прочиталась"?
...
пока вижу только, что мощность лазера не на 100% (225 и 1000 - разные настройки), а так особых "ошибок" нет, должно работать, если вы не "изменили" прошивку  и не переназначили вывод управления лазером.
Если у вас работает TTL-управление (есть поддержка PWM), то ваш лазер можно зажигать/тушить как параметром S, так и  M3/M4/M5.

Maxmgl, 2018-04-26:
С лазером проблема решена, надо было ставить настройку S1000, у нас 255 было.
Теперь осталось решить проблему с передвижением лазера ручном режиме. Стандартные кнопки программы посылают не верные команды типа $J=G91X0Y1F10, гравер не понимает.
при этом если ввести X0Y1 - все работает

Arhimed-Ru, 2018-04-26:
Maxmgl
Если  при  S1000 заработало, а с  S255 не работало, то почему в настройках не отметил, что  нет поддержки PWM?
И как это может быть, что не понимает G91X0Y1F10... ? версия GRBL машины "прочиталась"?
В крайнем случае можешь написать свои команды - создать свои кнопки

rockman256, 2018-04-27:
Есть ли конвертер г-кода для прошивки марлина? А то разраб не отвечает по добавлению фичи поддержки марлина. 
Код который сейчас генерируется марлин понимает только когда делается векторная обводка, в остальное время - лазер не включается (поддержки ттл нет).

closed 2018-05-31 by arkypita

Alexandr20011984, 2019-04-20:
Здравствуйте помогите нрбл почему то нет панели быстрого доступа, что бы менять мощность скорость и тд..как ее добавить?