arkypita/LaserGRBL #341: Two Steps in one Session.

herbiro, 2018-04-25:
Hi. Can you make a Option to engrave 2 Files (in my case: 1. Step: Line to Line & 2. Step vectoring without filling).

They are two .nc Files. Is it possible?

herbiro, 2018-04-25:
Oh. Perfect. Thank you.
Another question. Can you make a counter of pieces? 
Or:
Now it is how often the laser should engrave the Code. Can you make a Windows Like this: "Finish. It was number X from XXX pieces. Engrave the next?" Yes/Nö

Thank you.

martinisonline, 2018-04-25:
Good to know this new version add the ability to add more than one "job" at a time. This is really important when we need to add more than one "effect" to the same image.

Now, to be perfect, is the ability to edit/fine tunning any of the "jobs" that were added.

A simple menu with all the jobs already loaded and the ability to edit/delete the details of every one.

Thank you, and keep the outstanding work.

arkypita, 2018-04-26:
> Now, to be perfect, is the ability to edit/fine tunning any of the "jobs" that were added.
> A simple menu with all the jobs already loaded and the ability to edit/delete the details of every one.

Simple to say but not to do.

closed 2018-04-26 by arkypita