arkypita/LaserGRBL #342: Bugged version v2.8.44

arkypita, 2018-04-25:
I have released a new version called v2.8.44 (immediatly removed) that has an issue with translation.
This is because I have compiled it with Visual Studio 2017 that seem to have a problem with localization.

https://developercommunity.visualstudio.com/content/problem/63772/winforms-resx-localization-does-not-work-in-vs-201.html

Also this version does not do any update so user with 2.8.44 are strongly invited to revert to latest stable version: https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.42

Arhimed-Ru, 2018-04-26:
when will this problem be solved?
or need the old compiler?

arkypita, 2018-04-26:
For what I see if I compile LaserGRBL at home, where I have VS2017 freshly installed, I have an Issue with the translation (only english is working) but if I compile the same code with VS2017 at my office PC the issue with the translation disappear.

This is the version installed at office:

[image]

arkypita, 2018-04-26:
later I will check the version I have at home

Arhimed-Ru, 2018-04-26:
I have VS2017 freshly installed, I have an Issue with the translation (only english is working)
[image]

Arhimed-Ru, 2018-04-30:
I do not understand these compilations very well. 
but...
I think that the  probably the cause of the problem is not in the visual studio
I think that the new version of the framework (4.7) does not support your old localization language.
You need to update the localization files or use the old compilation framework (3.5 ... or older 4.7)

closed 2018-05-31 by arkypita