arkypita/LaserGRBL #343: Engraving

sabineh1, 2018-04-27:
Hello, I can not get a full path, I explain myself, I want to engrave a text but instead of just getting a path, a line, grbl engraving the outline of the letter, how to burn normal writing.
Thank you soo much

arkypita, 2018-05-31:
Already in #64

closed 2018-05-31 by arkypita

arkypita, 2018-10-21:
Hi all!

I have released a new version of LaserGRBL that include **centerline vectorization**.
This feature was requested by a lot of user and complete LaserGRBL with a "must have features" for all who need a simple way to engrave images.

This release was released as a pre-release so it is not automatically updated. 
To those who want to test it I recommend downloading the installer from: [here](https://github.com/arkypita/LaserGRBL/releases/download/v3.0.0/install.exe)