arkypita/LaserGRBL #345: Better overview of the change of the feed data, laser reading

Xillius178, 2018-04-27:
Have spent several hours with the program again, maybe a good idea:

The small message box to override the speed performance and rapid traverse our laser power.
Can you Move it permanently to the left side.

The Gcode rattles through the textbox so fast anyway, i can not track that.
In addition, most people have larger screens because it will not be a problem.

For this you may need a reset button to reset the parameters, or
a "right click" on the respective controller and he resets to the original value. so you can reset each controller individually.

An example, I have new wood that I would like to engrave.
Set the laser to S1000 (full power) now I can use the laser power regulator to set the needed on the wood during a test,
but when I'm with a mouse for a while somewhere else, the little window is gone again, but then maybe I wanted to vary the speed ....
So I have to click on it again and again.
if it were permanently integrated, that was much faster.

Just an idea from me ...

Greetings and nice weekend,

Xillius

Now:
[image]

Future??
[image]

kyroncfel, 2018-04-28:
Hi Xillius178,
Can you guide me on how you set the workspace as seen in the images. I will like my y axis to run in -y space as my homing is set at top left corner of machine

Xillius178, 2018-04-28:
Hello kyroncfel, it's not that hard.
You have to set the axes on the GRBL controller
As you like it
Stepper motor turn the cable over once or turn a spool (turn 2 wires)

I only changed the cables to the stepper motor.
Do you see a mirror image on engraving, then
change GRBL Config parameter 3 accordingly.

Then you need a program where you can determine the zero position.

I use Vectric Aspire 9 (with GRBL postprocessors)
[image]

But only because I also use it for my CNC milling machine.

There are certainly cheaper CAM programs with which this is also possible.

So it's not just the GRBL settings.

Hope I could help a little.

Regards, Xillius

kyroncfel, 2018-05-03:
Thank you for the reply and help. I will try and see if I can get in to work in lasergrbl and post back when I find a solution as my main intention is to do everything in one program.

arkypita, 2018-05-31:
Hi @Xillius178 

> For this you may need a reset button to reset the parameters, or
> a "right click" on the respective controller and he resets to the original value. so you can reset each controller individually.

I don't like the idea to have overrides permanent on the UI, but I understand the request. Maybe I can put an option on it. Instead I appreciate the idea of a "fast way" to reset it to the default, maybe with a double click or a right click action.

closed 2018-05-31 by arkypita

arkypita, 2018-06-12:
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.48