arkypita/LaserGRBL #346: customizable "Laser On" and "Laser Off" gcode commands?

beamzer, 2018-05-02:
In the [Target Image size and Options](http://lasergrbl.com/en/usage/raster-image-import/target-image-size-and-laser-options/) there are drop-down boxes for the Laser On and Off gcodes. 
I use LaserGRBL to control my EleksDraw (much better than EleksCam)  and the standard M3 and M5 codes work. 
However, the servo that lifts and lowers the pen gets steered to the maximal positions. But for the pen to lift it only needs 10 to 20mm. This would be possible if i could configure these commands. Pen Up could be "M3 S20" and Pen Down "M3 S40". That way it would be faster, make less noise and there would be less wear on the servo.
Is it possible make these configurable? (the drop-down boxes are already there ;).

arkypita, 2018-05-07:
already in development roadmap

arkypita, 2018-05-07:
This is not so simple as "the drop-down boxes are already there" because put them in generated gcode is not the only one thing to do... you have also to recognize them in gcode parser for drawing a correct image preview. Non standard, multi-command and custom "on off" could be more difficult to recognize

closed 2018-05-07 by arkypita

beamzer, 2018-05-07:
Ok, didn't think of that. But it's good to hear that it is allready in the roadmap.
Thanks for providing this awesome software.