arkypita/LaserGRBL #350: LaserGRBL Skips last but one command.

StuartB4, 2018-05-05:
Not sure if this is an issue or something that could be easily fixed.

I add $SLP command to the very end of a gcode file so that the stepper motors are 
switched off at the end of the job. 
It works fine and the motors can easily be moved by hand as they are no longer powered up.

Usually the last line of the gcdoe is always G0 X0 Y0, and at the end of the job the laser goes back to the home position.

However, when I add the $SLP command after G0 X0 Y0 the G0 X0 Y0 is ignored and the laser stops exactly at the end of the last line it burned, then it goes to sleep.
So it does'nt run the G0 X0 Y0 command but skips to the $SLP command.

Any ideas why would be greatly appreciated.

[image]

arkypita, 2018-05-07:
I think $SLP act as an immediate command at arduino side, so it is not queued into grbl move buffer but executed immediately

StuartB4, 2018-05-07:
Ok, but why does the laser not go to the home position when the burning has finished, it just stops at the end of the last line it burns.

arkypita, 2018-05-07:
try sending this command manually, enqueuing them as fast as you can so the command is not completely executed when you type the last one $SLP

F50
G1 X10 Y10
G0 X0 Y0
$SLP

[image]

As you can see grbl return an error (you cannot send $SLP while moving). This means that grbl manage the SLP command at the same time it receive, and not when the previous command is received. So, you cannot use $SLP in a queue of command but only perform it manually when the job ended

I don't know why in a file streaming operation it act like this (without sending an error but stopping motors in a wrong pos) however this look like an issue of grbl, not LaserGRBL so this is not the right place where ask

arkypita, 2018-05-07:
already tried putting a G4 delay between G0X0Y0 and $SLP ?

closed 2018-05-07 by arkypita